df8oe/UHSDR #347: Request: store peak filter and notch filter setting with power off

DG9BFC, 2016-05-05:
when i have set peak filter to 700hz ... or the manual notch to any setting ... i would like to have the same setting after power off / on again ... so can we make those settings stored with power off??

73 sigi

dl2ki, 2016-05-07:
Hi,

Why just 700Hz?

73, Wolfgang

DG9BFC, 2016-05-08:
many digital modes in cw mode has 700hz as default

i also have set my hdsdr offset (for cw mode only) to that 700hz

and ... many users prefer a lower tonne for cw (some even go down to 
400hz or so!!)

....

the reason for asking for 700 instead of 750 is ... cause it is a common 
value for cw offset

many conventional rig also have those 700hz as default for the cw offset

.....

now clear??

greetz

sigi dg9bfc

DG9BFC, 2016-05-27:
Thank you programmers for adding :-)

closed 2016-05-27 by DG9BFC