df8oe/UHSDR #348: Request : colour of s-meter bar should be configurable

DL3NCW, 2016-05-06:
For better view it should be possible to change the colour of the s-meter bar
like it is possible for spectrum scope and waterfall

73 de Joachim DL3NCW

S54AM, 2016-05-16:
Also other bars (PWR, SWR, ...), should be added into this request.

73, Marko, S54AM

df8oe, 2016-11-09:
All such "improvements" will blow up configuration so that a newcomer cannot configure his mcHF without RTFM (do you want to write such a document?) We already do have >100 configurations and only neccessary configs should be added.

S54AM, 2016-11-09:
Hi Andreas,

Yes, I agree. We have too much improvements but somebody needs to track,
filter and collect them, because all of them are not really usefull and the
best one. And newcomers, as you mentioned, are really confused. There is
too much topics in conversations section (I mean in mcHF Yahoo group).
Maybe we should divide project by topics on forum separated to sections.
When I was building the radio I spent a lot of hours on RF final stage and
my T7 solution based on almost dosen of practical tests and measurement. So
we, let say big or small developers, should have our specific interest
"room".
Andreas, you are the strongest engine of this great project and thanks for
your big effort. Because your enthusiasm improved the radio almost to the
top level which existed hardware can allow.
I am permanently a broad up to May 2017 and also out of my shack, so my
contribution is currently limited. Every couple of days I check situation
on the mcHF forums and your github, but If you want you can use my T7
solution described in the latest KA7OEI modifications for further more
useful manual.

https://xa.yimg.com/df/M0NKA-mcHF/mcHF_board_modifications_20150928.pdf?token=WpPvLn9k633B586hV4kxIonAmU6vIfWqAGXEyQm6Qs2wG6yH7WlND3dics0LLo26rY1Brn9_P0OtXCS1F8BykvKR1kjcNaPPDEW7OOg96I_63j298UO-B_xuqBbgTJxJvRMY-nfQZYRBxZ8&type=download

73, Marko, S54AM

df8oe, 2017-01-06:
implemented.

closed 2017-01-06 by df8oe