df8oe/UHSDR #360: Request: AGC setting on display

DG9BFC, 2016-05-12:
we now have 3 values in a row ... so far so good (i like the new look ... thumbs up)
would it be posible to add agc setting in display ... (slow med fast custom and if selected custom then display the number from menue and make it changeable with encoder)

what do you think??

db4ple, 2016-05-13:
Hi Sigi,

yes. This is possible. See comments for #361. However, we should collect ideas for use of these boxes. With the current layout up to 12 boxes are possible in future but getting more than 12 will be hard to achieve. So we should be careful here.

DG9BFC, 2016-05-13:
12 boxes ??? ... yes then we could add agc setting

if using of noiseblanker lowers the agc value (means desensing rig) then 
selecting a faster agc does help (so sensitivity is faster back in a 
higher level) ... that is why i would like to have it on the display

....

using dsp (nr) does change the tone somewhat ... and that is why i would 
like to have bass and treble on display (permanent) ... and setting 
should be stored with power off!!!

73

sigi

closed 2016-11-09 by df8oe