df8oe/UHSDR #361: new display settings waste room

DG9BFC, 2016-05-12:
as can be seen at treble and bass box in one row we can put 2 values with description in one row
... that means:
in the same area where we now have six boxes with six values ... we could display 8 values ... or if you count bass and treble together with the 6 boxes there would be room for TEN VALUES that are configurable for the user ...
so the new box sized look really does waste room

cause we have so many functions that are configurable by the user ... it would be better to make 5 or six rows with two values (like can be seen in the example of bas and treble) ... means 10-12 configurable settings ... enough to make bass and treble permanent on screen ... plus add agc setting also (and what else???) and still have some room left below for manual notch or peak filter

what about the possible setting of audio (speaker gain) on left encoder and bass and treble on the other two
... (in am then it would act like a normal kitchen radio hi hi) ...

means that with the button you can switch the different active boxes and encoders like we have now ... but bass and treble are not on same encoder ... so with a clever setting of the active boxes and encoder you have then volume bass and trebl on the three encoders

encoder 1 with below button 
afg cmp agc

encoder 2 

rfg dsp bass

encoder 3

rit mic gain treble

.... so you can have volume bass and treble as one setup (maybe with power on as first setting??) ... what do you think??

(just an idea) ....

73 dg9bfc sigi

db4ple, 2016-05-13:
Hello Sigi,

there are multiple reason why we switched the layout. 

The main one is that now each encoder has its  own column. So left most boxes are encoder one, middle encoder two, right encoder three. Bass and treble boxes will be matched to that model in future. Exactly as you proposed. However, this requires some changes to the existing code which we do not want to make right now as we are in a bug fixing mode.

Next reason is that more square of same area size are much easier to hit in touch actions (and also better and more precisely identifiable in code, since touch screen is only 60x60 touch points).

The space is wasted for a good cause, we believe.

db4ple, 2016-05-13:
Forgot to mention: Planned feature: Touching on a box will allow you to enable the change that setting with the corresponding encoder. That will speed up things if we have more than 2 boxes per encoder.
Buttons below boxes will still work as before.

db4ple, 2016-05-13:
Hi Sigi,

at the moment we have implemented a different approach for Bass and Treble: it is activated the same way as before but instead of being displayed below the Encoder 2 column is switched to Bass&Treble. Let's see how this works out.

dl2ki, 2016-05-13:
Hi,
the solution now found for BAS / TRB is well done. Less is often more.

A broad line below the TCXO display would give a better visual separation of the areas.

The text box "WATERFALL DISPLAY" and "SPECTRUM SCOPE" could possibly also be omitted and be replaced by a wide line.

73, Wolfgang

df8oe, 2016-05-13:
Line WATERFALL / SCOPE is used to oreintate for touchfunctions "magnify / toggle wf & scope" so it must remain at the moment.We are currently working on a more consistent ui and will find solutions which take care of all aspects.

So I will close this for the moment.

vy 73
Andreas

closed 2016-05-13 by df8oe

DG9BFC, 2016-05-13:
ok ... box layout is for touch on these boxes to activate them (or use 
button below its corresponding encoder till box is active) .. ok on that

DG9BFC, 2016-05-13:
looks nice ... but if you could now put the treble to right encoder ... 
then we would have the "kitchen radio" setup

(volume bass and treble on the three encoders)

now you have only two functinons on encoder three but a lot more on 
second encoder

... if treble send to right encoder that would be better (in my view)

encoder left : af gain compressor and (maybe agc setting??)

mid encoder: bass rfgain dsp

right encoder: treble rit and inputselector(inputgain)

....

means treble jumps to right encoder (more functions on right and less 
functions in mid encoder)

.....

so mid encoder is not so overloaded ... and with power on we can have 
volume bass and treble as default

....

what do you think??

73

sigi

df8oe, 2016-05-13:
It is mostly used as a transceiver not as a broadcast radio. And its concept IS a transceiver! Bass and treble are for "hifi fans". So this is not recommended - my point of view :)

vy 73
Andreas

DG9BFC, 2016-05-13:
it was just an idea

  ... no need to follow it

73

sigi

df8oe, 2016-05-13:
Of course there are thousands of possibilities how to handle xyz... We want to follow the idea of a logical and consistent organization of ui functions. There is already done some work - but much will op up by thinking about existing structure. Encoders and boxes are optically in corellation - that is the way we want to organize them. And bass/treble/notch/peak are all functions related to audio frequency settings - so they should be located at the same encoder... What about AGC e.g. is not thought about deeplyat the moment - will follow ;)

vy 73
Andreas

DG9BFC, 2016-05-13:
i have read that somebody used encoders with included push button and 
wired them to the three buttons below encoder ... nice idea ... right??

anything that we have seen during the last months was good ... all 
cosmetics ... and all internals (filters etc.)

....

so i am sure also the next ui changes will be very good after all 
.....even if there may come some "intermediate" versions that do not 
look so well or where a value is missing ...

the new box design looks nice ...

sure you will find a good solution for all needed values in the end ....

i was asking for storage of settings

bass treble and also peak filter and manual notch

can those four values be stored with power off??

so with next power on you have same values???

73

sigi