df8oe/UHSDR #368: Menu button

S54AM, 2016-05-14:
Menu should be accessible harder than now. Maybe with combination of two buttons, that guest users on radio wouldn`t be able to change settings. In this case, menu button should be used for other options. Maybe for locking all buttons (LOCK). 73, Marko, S54AM

DG9BFC, 2016-05-14:
we already have vfo lock ... what els should be locked additionally?? (maybe mode change added to vfo lock??)

S54AM, 2016-05-14:
Yes, I know about VFO locking which could stay separated, but I mean possibility for locking all other buttons. My main suggestion is harder access to menu.

DG9BFC, 2016-05-14:
and next step will be pin code lock?!?

not that bad idea for sensitive configuration settings to hide behind a 
pin code wall? ... with sensitive settings i mean such as iq correction 
or bias ...

all other things couild be configurable like we have now ...

dg9bfc

sigi

S54AM, 2016-05-14:
I agree about keylock, using soft keyboard, I suppose :).
Sorry that I were not clear enough. 
Suggestion one: Somehow coded access to menu to avoid "surprises" when you land radio for example (for all functions which are not accessible with buttons and encoders, if is possible). 
Suggestion two: Lock function for locking all buttons (except power).

S52CQ, 2016-05-20:
Please do not add lock feature with Touch screen. Not all of us have Touch screen enabled due to mechanical construction. Namely I have a transparent PVC for a window placed in the enclosure in front of the LCD.

But I agree with some sort of locking, maybe with a combination of pressed buttons.

73 de Jure, S52CQ

df8oe, 2016-05-20:
Hello Jure.

putting LCD behind PVC is like stopping your project at the recent development stage. That is your decision - but be sure that many new features will only be available via touchscreen. So you can rest at a stage you want to do this - but development will go on... And mcHF *is* a touchscreen receiver at all!

In the actual case we are not sure a menu locking is useful... But time when you get problems without touchscreen is approaching.

vy 73
Andreas

S52CQ, 2016-05-20:
Well Andreas, we say that each eyes have it's own painter. In the enclosure that I have the PVC means a nice finish, the enclosure looks professional. Without it it looks like a tooth with a big hole in it. To have a touch screen is a nice feature but not on such a small touch screen where a pen and a pair of glasses should be used. 
Those are main reasons why I will keep mcHF behind the development as you say. As such it's already nice piece of equipment that works great thanks to you guys.

I'll gladly buy you a beer (or two) in Friedrichshafen at the fair if we meet in person. I realize that a bier or two cannot be a payment for the work that you do, but at least I can try to show with it that I respect the work that you do for the mcHF community, although I may not agree with some of your decisions.

73 de Jure, S52CQ

df8oe, 2016-06-23:
May be some day we implement this. For now it has very low priority...

vy 73
Andreas

closed 2016-06-23 by df8oe

S54AM, 2016-06-23:
Hello Andreas,
Thanks for information. It will be great to meet you in Friedrichschafen this weekend. I am part of Slovenian ham delegation there.
73, Marko, S54AM

S52CQ, 2016-06-29:
Hi Andreas,

I was hoping that you will stop by at the S5 booth for a bier...

To continue the Menu lock feature, on the fair it has proven that Menu lock feature is needed. As you may know mine and Marko's mcHF were displayed at HAMRADIO last week and both stations menus were changed while we were not watching. I agree with you Andreas that this is not high priority because we don't take our mcHF to public very often, but it would nice to have this on TODO list.

BTW, many HAMs that did not know mcHF were very impressed with the radio and features that it offers. Also we got a visit from Eric, WA6HHQ, president of Elecraft and he stopped by mcHF playing with it. 
[image]

73 de Jure, S52CQ

df8oe, 2016-06-29:
Hello Jure,

nice to hear! I have not visited HamRadio this year - otherwise of course I would have visited you ;)

Button lock is a feature which produces many questions...

- a simple button combination is only a secure for unwanted changes - not for wanted ones. The combination is documented and so known by others

- creating a touch function like "connecting postiions with lines" like on smartphones is possible. But it is not a very simple solution, it must be user-definable and it takes time to deactivate it

- an easy solution would be pressing paddle-dah and paddle-dit simultaneously while press&hold another button. So if you unplug keyer no chance do deactivate :)

Again: describe what kind of lock you want

73
Andreas

S52CQ, 2016-06-30:
Hi Andreas,

OK, we might drink that beer at Electronica this November if you'll be there :)

Your idea of pressing dit/dah & holding another button is great to toggle MENU button.

Nice and simple solution!

73 de Jure, S52CQ