df8oe/UHSDR #378: new version sometimes hangs with no audio

DG9BFC, 2016-05-17:
mchf set to am ... nr engaged (not nr+notch! ... just nr!) ... turned vfo a bit up and down (searching over some radio stations)
then suddenly audio dropped away
and did not come back ...

tried to switch off nr ... changed modes ... nothing brought audio back (except power off/on)

73 sigi

db4ple, 2016-05-17:
Can you reproduce this? I.e. does it happen repeatedly?

DG9BFC, 2016-05-17:
yes ... had it three times than uploaded an older bin from 13th of may

to ckeck if it is also there ... no in 13th may version all is well (can 
tune as fast as i want an no hanging fw)

...

sadly i do not have version from 15th cause I accidently deleted it when 
downloading that newest one from today (forgot to rename the other 
version before downloading on usb key)

maybe andreas can send me pre-last version by mail so we can check where 
that EXACTLY started ...

  ... i had switched 2 tone on in menue (just to see its function) ... 
but can not say if the fault is coming from there ...

...

so if andreas would be so kind and send me the pre-last version i can 
check if that fault is from today ... or yesterday

greetz sigi dg9bfc

db4ple, 2016-05-17:
Sigi, try to use 1.0.0: https://github.com/df8oe/mchf-github/releases

DG9BFC, 2016-05-17:
seems as 100 does not have this problem ... (it does not hang when i 
tuned around ... bt have to do some more testing time) .....

i will have to reload the other version to test if then the fault is 
again there (so reproducable)

.....

73 sigi dg9bfc

db4ple, 2016-05-17:
Ok, this narrows it down to a few commits. I'll look into this tomorrow.

db4ple, 2016-05-17:
Do you have a CAT program running?

DG9BFC, 2016-05-17:
uups ... correction ... now it DOES hang!!

so 1.0.0 also does not work ... (hangs after some tuning)

...

details: am mode, noiseblanker 10, noise reduction 5 ... tuning around 
in 19m radio band (15-16 megs) ...

suddenly a "klercronk" sound and then ... silence

mode change does not help (it switches modes but no sound from speaker)

....

now i have to test older versions ... right??

...

DG9BFC, 2016-05-17:
no usb cable connected ... so not cat soft running

db4ple, 2016-05-17:
If it happens with 1.0.0 CAT should not be an issue, the only CAT change was made after 1.0.0
Still, I'll have to look into it tomorrow in more detail.

DG9BFC, 2016-05-17:
ok ... version 27.16 seems to work ok

27.17 , 27.18 ... and 1.0.0 hangs after some tuning around

...

witrh 27.16 i can tune around as fast as i can  ... turn the knob up or 
down ... no "klerkronk" and hang ...

(exactly soft does not hang ... cause tuning and spectrum does work ... 
simply no audio out)

....

greetz

sigi

DG9BFC, 2016-05-17:
no cat cable connected ...

...

it is between version 16 and 17 I guess

greetz

sigi

(off to bed now cause i have to be up early in the morning)

.....

db4ple, 2016-05-18:
Hello Sigi,

no change between 16 and 17 seems to be obviously related to tuning. Might have something to do with the NB and NR being enabled. Or being changed in value, since what was indeed change was the code around the handling of encoders which are used to control the NR and NB settings.

DG9BFC, 2016-05-22:
now fast tuning switches off the nr function ... so seems as we can close issue now :-)

closed 2016-05-22 by DG9BFC