df8oe/UHSDR #379: DSP NR engaged : No fast tuning

DL3NCW, 2016-05-17:
While DSP is set to NR no fast tuning is possible despite the
fact that it is activated through touchscreen.

db4ple, 2016-05-17:
Haven't tested it, but most likely the NR settings are causing to much cpu load and this causes the encoder to be queried to slowly to detect the tuning speed correctly. NR is a first class performance eater, unfortunately. There is for sure no code which  disables the dynamic tuning if NR is turn  on.
Could you please also state the version of the firmware and the type of display (SPI vs. parallel) you are using? Thank you.

DL3NCW, 2016-05-17:
I am using daily from May 17 morning.
Display is in SPI mode.
If it is so, can't you change the step of DSP from DSP OFF - NR - DSP A-NOTCH... to
DSP OFF - DSP A-Notch - NR...
That would prevent  switching NR on if G2 button is pressed accidentialy

DG9BFC, 2016-05-22:
seems as the programmers now switch nr off when using fast tuning (or bigger tuning steps) ...
when i tune in BIG steps (means 100kc) or have fast tune swiztched on (with tapping on stepsize button on the touch pad)  then nr is switched off ... and to reengange i have to turn the af gain a bit (up or down) ... turning the encoder for nr does change value in nr box but does not switch the nr function back on (if switched off by fast tune) ....

DG9BFC, 2016-05-22:
so ... we can close this issue i guess

closed 2016-05-22 by df8oe