df8oe/UHSDR #382: Mode switching in the 60m-band

dl2ki, 2016-05-21:
Hi,

my settings: 

- LSB / USB Auto Select = ON
- CW Freq. Offset = LSB SHIFT

20m-band USB: press the "Mode" button switches correctly to between USB and CW-L.

60m-band USB: press the "Mode" button turns first to LSB and then on CW-L. Then switches between LSB and CW-L. USB / CW-L would be correct.

73, Wolfgang

df8oe, 2016-05-22:
Hi Wolfgang,

when I togggle on 60m my mcHF switches between LSB and CW-L and on 20m it switches between USB and CW-U. Both is correct... My setting for CW Freq. Offset is "AUT USB/LSB". When I set it to  "LSB SHIFT" it always uses CW-L (on the band with normal use "USB", too). That is correct. And on 60m normal mode is LSB - so your reported behaviour is ok. USB and 60m would be a fault.

So I do not see an issue.

vy 73
Andreas

db4ple, 2016-05-22:
Hi Andreas,
I have no idea about the correct mode on 60m CW but the code does make optionally an exception for 60m when operating SSB and  the special 60m USB case is configured via menu. 

We do not care about this special setting in CW on 60. But we could. 

Danilo

dl2ki, 2016-05-22:
Hi,
far as I know, on the 60m band the prefered voice operation is in upper sideband (USB).

> Upper sideband should be used for SSB operation at 5MHz to preserve compatibility with other services.

The CW mode (LSB) on all bands is a correct setting and is not dependent on SSB.

73, Wolfgang

df8oe, 2016-05-22:
There are two different informations:

- frequencies lower than 10 MHz == LSB, frequencies higher than 10 MHz == USB.
- in the United States USB is preferred for 60m.

So these two are not solvable. Clint decides to follow the first statement...

vy 73
Andreas

dl2ki, 2016-05-22:
Hi Andreas,

on the 60m band you can hear stations usually in USB. So I did some research, and found this information:

- [http://rsgb.org/main/operating/band-plans/hf/5mhz/](url)
- [https://en.wikipedia.org/wiki/60-meter_band](url)
- [http://www.arrl.org/60-meter-faq](url)
- [http://www.arrl.org/files/file/Regulatory/Recommended_Practices_Version_6_5.pdf](url)
- [http://folk.uio.no/tomvs/la4ln/Worldwide%2060%20m%20frequency%20chart_v10.pdf](url)

So I think the USB right mode for SSB.

73, Wolfgang

db4ple, 2016-05-22:
Hi,

I propose to do it like  this: we on CW Auto we use the settings in the SSB menu to set 60m behavior. So if the user decides to opt for 60m and USB as auto sideband for SSB, we use the same logic for CW.
Otherwise below 10 LSB above USB. Nor no auto.

Consequence of that is: If the user decides to not do SSB Auto, but auto CW the CW code will not be able to do CW 60m special handling (since the SSB settings is not available in this case).

The other solution would be to make separate setting for CW, I am no friend of that. More place where you set similar things make it more complicated. 
I assume that you would have auto enabled for both or not at all.

Other other hand, as a no CW guy, I don't get the point of importance of  sideband in CW anyway. For decoding no difference, frequency of the signal will be same no matter which sideband you use  to listen to it. 

To conclude from my point of view: We either  leave it as is or change it according to my proposal or have  more voices which demand a different, more complex solution.

That makes most sense to me.

df8oe, 2016-05-22:
USB and LSB decision is made in a very easy way in firmware: all below 10 MHz uses LSB as default. It is possible to program band specific modes but this will be work because of complete concept must be changed.

I suppose be rest at the existing oncept and toggle USB/LSB as we want...

vy 73
Andreas

db4ple, 2016-05-22:
Hello Wolfgang,
I assume you have Auto USB/LSB set to 60m in the settings, right? And you are  now wondering, why the 60m are handled correctly in SSB but not in CW.

Right?
Danilo

dl2ki, 2016-05-22:
Hi Danilo,

in my first post I described my menu settings:

- LSB / USB Auto Select = ON 
- CW Freq. Offset = LSB SHIFT

CW-Mode settings and SSB-Mode settings are two DIFFERENT things. Most transceivers use for CW only one sideband. So "Auto USB/LSB" is a setting for SSB Modes.

The commercial manufacturers handle this differently.
Kenwood CW RX is always USB, Elecraft always LSB. ICOM (LSB) and Yaesu (USB). Regardless of the band! 

I use "LSB SHIFT", as this allows the frequency display coincides with my K3 (CW = LSB). My Kenwood TS-570SG I have to adjust to CW-R, as Kenwood used USB-CW. The RX-frequencies otherwise differ by the amount of sidetone frequency.

If someone has transceivers of from different manufacturers at home, he can check it once together with the mcHF. Best with a time signal transmitter, for example WWVH, and Fldigi or an audio spectrum analyzer.

The manuals of the various manufacturers give to the CW mode also information.

73, Wolfgang
DL2KI

db4ple, 2016-05-22:
Hi Wolfgang,

To get the desired 60m handling you simply have to set "LSB / USB Auto Select" to "60m"

I think we should close this issue now. We leave CW and SSB handling as it is currently, since it takes care of your wishes (once you select the right menu settings that is).

Agree?

dl2ki, 2016-05-22:
Hi Danilo,

the menu setting "USB 60m" I have not seen so far. SORRY.

73, Wolfgang

closed 2016-05-22 by dl2ki