df8oe/UHSDR #384: Request: Temporary frequency  memory

dm6tt, 2016-05-24:
It would be nice to have a possibility to store an interesting frequency when 'scanning' the band manually. This can be done as round robin with about 10 memories places.
Pressing one button enables the temp. Memory. One encoder can be used to select the memory for reading. This functionality is similar to the TS-850.
I would suggest doing in band independly.
Since it is temporary it need not stored in the eeprom.

It would be very helpful to easily reach the interesting stations that were heard while scanning the bands.

73, Marcus, DM6TT

db4ple, 2016-05-25:
The function itself is not a big deal. But have you an idea which of the existing buttons should activate this function when? This is where the fun starts...

df8oe, 2016-05-25:
The only way to implement this in an intuitive manner is creating a touchscreen ui. If we do this we can implement much more nice features and they can be reached in a self-explaining way ;)

vy 73
Andreas

db4ple, 2016-05-25:
To give an more or less complete overview, I added the key assignments to the Wiki page (Thanks Andreas for the initial list): https://github.com/df8oe/mchf-github/wiki/Operating-Manual

BTW, even with touchscreen: It is not endless space. But some is there, I agree.

df8oe, 2016-06-19:
I think you can use VFO-B. Closed for now.

closed 2016-06-19 by df8oe