df8oe/UHSDR #387: new display layout

IZ6MAF, 2016-05-27:
It would be nice if you could enlarge the spectrum/Wfall on the display with a new layout as shown in the attached photo montage
Best Regards 
Paolo IZ6MAF
[image]

DG9BFC, 2016-05-28:
nice picture ... well done
but andreas will moan that his grey bar is gone and he uses that grey bar for two touch functions (zoom in/out and toggle spectrum/waterfall) ... so we then have to find 2 areas for these touch functions (and/or two button press or longer press functions for non touchpad user)

by the way we are taling about gui ... what about adding a second zoom ratio (times 4 instead of times 2) ... then you can see the signal (and only that) on the screen ...

db4ple, 2016-05-28:
Hi Paolo et al,

Looks cool.
 
Some points w.r.t. 4x:
However, I will have to tell you a secret: Right now, if you do magnification, you do not see more you see just half of the data twice as wide. Question is if this gives real value if same is repeated.

To improve: Processing half the bandwidth data with FFT (in order to have twice or four times the resolution)  needs changes to the code (I don't know how much). Once this is done that, it is probably fairly easy to also have it running for 4x mag. But I am not the FFT expert here.

Processing the full set of data with twice as many FFT points is probably too timeconsuming (don't know since I haven't measure, but for sure will take twice as much time as now).

Some points w.r.t. Wider Spectrum Display: 
I think extending the Scope by 25% (256 -> 320) is feasible in terms of FFT effort and also memory usage. 

However, the waterfall is a different story. Waterfall display eats right now already about 60k out of 192k of RAM. The whole rest of the SW eats another 64k (!, around 13k of this used by Snap Carrier alone) right now and new modes such as FreeDV need also lots of memory. Extending by 25% would required roughly ~20k more taking into account the larger FFT memory I would say. 
88

Relocating the UI elements is a no-brainer.

This is no judgement, just background information.
 
Now the question of all questions to all of us: Is this worth it?

df8oe, 2016-05-28:
Hmmm....
LPF box now would conflict with frequency digits for 144, 430 and 1200 MHz. You wil not get more information out of spectrum/WF - but drawing will take more time (LCD a bit slower) and Ram consumption increases.

May be it "looks better" for some eyes - but technical it only has misadvantages... So I do not want to change there anything.

vy 73
Andreas

DD4WH, 2016-05-29:
Hi guys!

@Paolo: I really like the idea of your setup. However, making the spectrum display wider than 256 pixels calls for major changes in the code as Danilo already stated.
Processing half the bandwidth is not possible. FFT takes all the bandwidth (in fact, the sample rate bandwidth = 48kHz) and transforms time domain into frequency domain. That produces 256 usable data values with a 256bin-FFT. Frequency resolution is therefore 48000/256 = 187.5Hz = 1 pixel. If we now display 24kHz, the resolution remains the same (now: 128 values for 256 pixels), so two neighbouring pixels have the same value. With magnification 4x (64 values for 256 pixels), four pixels have the same value. Only solution to this is using a 512bin-FFT (only available FFTs are 256, 512, 1024, 2048 etc.). then we would have 512 values for 48kHz and 256 values for 24kHz displayed bandwidth (= 94Hz bin width resolution per value). (1024bin-FFT would be needed for 4x magnification). This doubles/quadruples the processing effort and the memory effort. This would be not a severe problem for the spectrum display, but it is definitely not possible for waterfall (not enough RAM! Believe me, I tried it for 512bin-FFT, not to speak of 1024bin-FFT ;-)). 
So: this is worthwile trying out for spectrum display only, but would require some major changes in the code AND it would surely make the mcHF a lot slower in 2x magnification mode and would definitely make DSP NR and other processor-consuming goodies impossible, when in 4x magnification mode. I see potential conflicts when we implement a larger FFT for spectrum display if that compromises other (more important!?) functions of the mcHF.
If we do not change the FFT size, making the spectrum display wider would be useless, because at the moment it shows ALL the detail that is in the FFT data when in normal mode and half of the data that is there when in 2x-mag mode. I do not use 2x mag mode, because you do not have an advantage from that, because you do not see more detail.
73 de Frank

df8oe, 2016-05-30:
So I think we can close this feature wish...

vy 73
Andreas

closed 2016-05-30 by df8oe