df8oe/UHSDR #389: Sidetone out on USB & Line-out

Rescue9, 2016-06-01:
Now that CW sidetone is generated without transmitting, the ability to have sidetone output through the USB audio and Line-out is important. The ability to hear a tone generated by the radio is especially useful for operators who are using the mcHF to generate code during practice sessions. Being able to directly interface this sidetone into a computer or android device enables the operator to have the tone generated via Skype (used by CWOps training personnel) and other meeting software.

Rescue9, 2016-06-01:
Previous discussion on this issue can be found in part on the other sidetone issue. https://github.com/df8oe/mchf-github/issues/332

Rescue9, 2016-06-01:
After reviewing all information, I believe the ability to have sidetone output through the audio system simply needs to be an option in the CW menu settings. IMHO there doesn't need to be a button presss linked to this as it will be used primarily by students, or beginning cw operators, and will likely affect only a small percentage of mcHF users.

df8oe, 2016-06-23:
May be some times we will implement this - for now it has very low peiority...

vy 73
Andreas

closed 2016-06-23 by df8oe

PA3CCE, 2016-06-23:
Op 23-6-2016 om 11:52 schreef DF8OE:
>
> May be some times we will implement this - for now it has very low 
> peiority...
>
I do NOT agree on the remark of LY2BOK regarding the dBm/Hz display, I 
think this a very useful feature.
Who cares about these few digits when you don't use them?

Jos (PA3CCE)

df8oe, 2016-06-23:
... I do not agree to LY2BOK, too. And I do not agree with some others of his feature wishes and opinions... But we all are free to write feature requests and tell our opinion regarding to mcHF - o it's ok for me.

vy73
Andreas

PA3CCE, 2016-06-23:
> ... I do not agree to LY2BOK, too. And I do not agree with some others 
> of his feature wishes and opinions... But we all are free to write 
> feature requests and tell our opinion regarding to mcHF - o it's ok 
> for me.
>
> I fully aree to the above , dont misunderstand.

Jos

df8oe, 2016-06-23:
I know Jos - I never thought other way...

vy 73
Andreas