df8oe/UHSDR #408: Volume level in USB Audio Digital(not DIG IQ)

bh3nvn, 2016-06-17:
How can I change the USB TX Audio level and RX Audio level on mcHF?

1.When I set the TX Aduio Source to "DIGITAL",I used the knob3 to change the TX gain number,but the output power did not change.But the output power can be controled by software(HRD,WSJT-X, etc) or  Windows Sound Mixer.
And in addition that  the TX gain number maintain 12 when powered on until push some button(like"MENU").

2.In JT65 or PSK31 mode, sometimes I want to reduce the RF Gain and increase the RX Audio volume,this can help keep a strong signal from wiping out the weaker ones.But the RX Audio volume do not increase when increase the microphon Level(mcHF USB interface) on windows10(Windows Sound Mixer).The RX Codec Gain in MENU is useless too.
So,how can I increase the USB RX Audio volume level on mcHF?

db4ple, 2016-06-17:
Indeed does the USB code implements Audio Volume control via for the mcHF, i.e. you are controlling the rx and tx gain values using the Windows volume control. Unfortunately the firmware does not make use of the  tx_gain value for digital input. Well, don't remember why I didn't implement that. Surely it is not strictly necessary, but you then right to ask, why then dispaly and allow to change a value with no meaning at all. Good question. Should fix that at some point.
The menu thing  is an "graphical update issue". The USB interface changes the value but value display is only update unconditionally at some points (e.g. when you switch on/off menu mode).
You can play around with the Windows volume level and will see the value changes (but of course nothing happens).

RX gain is different, you cannot control that using the mcHF knobs but this value is being actively used. You can see that if you change the Windows settings on Windows 7, this will impact the volume of the transmitted audio signal directly (at least that is how I remember it). On Windows 10 you're right, it does not work. Don't know why yet.

I suggest to leave that issue open, and some times later I'll look into that.

db4ple, 2016-08-12:
Some progress: I tried to connect the gain settings of the digital input with the actual gain code. It does work (no surprise here). However, we have to decide if we actual want to be able to control the gain via PC and mcHF or only PC or only mcHF. Difficulty with the PC&MCHF approach is, that the mcHF cannot report back its internal set value. There is no back channel for to the PC  settings implemented. Only mcHF or only PC are no problem but which one is difficult to decide.

df8oe, 2016-08-12:
My opinion:
only adjustment via PC is done. So you can level each application with its individual settings. There is no need to do TX adjustments via mcHF. So it is neccessary to eliminate settings display in DIG and DIQ nput mode from screen.

df8oe, 2016-12-09:
This is now finished. You cannot set level at mcHF when using USB audio. It is fixed at maximum level.

closed 2016-12-09 by df8oe