df8oe/UHSDR #416: dBm

oz8gw, 2016-06-18:
Hi Andreas.
Good idea with the dBm-display.
A bug: The choice dBm or dBm/Hz is not saved in the menu.
I wonder why S9 does not match -73 dB
Vy 73 de OZ8GW, Leif.

DD4WH, 2016-06-18:
Hi Leif,
for this first trial version I thought it would not be necessary to save the settings. So that is not a bug, but a "feature" ;-). Maybe we can save that, but as you know, EEPROM space is scarce . . .

I spent some time calibrating the dBm display by comparing the mcHF dBm display with a direct sampling SDR dBm display. It has to be further fine-tuned, I know and will do that in the next weeks. My experience is, that the dBm display is about +-3-5dBm accurate.

The S-Meter of the mcHF is a totally different story and it also has a totally different basis of the calculation. I did not touch the S-Meter at all, because in discussions in the programmer team there was strong support for leaving the S-Meter as it is, although everyone is aware that the S-Meter does not show reliable S-measurements. (the mcHF has that in common with most commercial transceivers: see https://github.com/df8oe/mchf-github/wiki/S-Meter-&-dBm-Hz-display). 

So you are perfectly right: S9 on the S-Meter does not mean -73dBm in reality. And exactly that was my motivation to set up the dBm display ;-). Hopefully a real -73dBm signal shows up on the dBm display as (about) -73dBm on the dBm display ;-).

Go ahead and test the accuracy of the display, I am always interested in measurements that help optimizing the mcHF!

73 de Frank

df8oe, 2016-06-19:
moved to "closed"

closed 2016-06-19 by df8oe