df8oe/UHSDR #437: software error

ly2bok, 2016-07-04:
Dear programmers! When testing the program from 3.07.6016 17:27:34 problem. When the radio is turned on, there are situations that are not completely suppressed by the mirror channel. To eliminate this effect it is necessary to turn off the radio and then re-enable it. Sometimes you this was done several times to the radio started to work normally. In all likelihood it occurs when the pogramny full of failure. This effect it is observed on multiple devices. Please pay attention to it.
LY2BOK
LY3BD

db4ple, 2016-07-04:
Hi,

this problem is known as the "TwinPeaks" phenomenon. We have no clear idea why it happens. It is a problem cause in/or close to the WM Audio Codec since the IQ I2S stream has an offset sometimes. Restarting the data stream from the Codec solves the problem.

There is a function available in the Standard Menu "Restart I2S" which should be used to reset the codec. In most cases, this will fix the issue without restart. Same function is also run if you have touch enabled  and press the "DSP" mode area on the left side.

db4ple, 2016-07-04:
This is not a direct software error, since then it would be reproducible. The problem has been around for very long, but for  whatever reason, we are seeing it more often with more recent firmwares.

ly2bok, 2016-07-04:
The problem is  OK.  We see.
Thanks you
LY3BD , LY2BOK

closed 2016-07-04 by ly2bok