df8oe/UHSDR #450: offset (if shift) does not work correct and tuning very slow

DG9BFC, 2016-07-13:
tx/ rx frequency xlate set to -12 kc but it receives at +12 kc ... plus the readout of the vfo does not match correct ...

and ... tuning is now very slow (it stops several times when you turn the button) ... and sometimes switching to red digits

df8oe, 2016-07-13:
Hmmmm......

IQ adjustment works in actual version identical as it worked since "mchf Abgleich Party in Sudwalde" a month ago. Same with VFO setting - I cannot find any difference to that version (andall ones from then to now). Tuning is as fast as it was - I never get red digits. I tested with "dynamic tuning on" an step size "100Hz". If you choose greater steps red digits are normal because of SI570 must do large tuning steps if you change frequency too fast.

DD4WH, 2016-07-13:
Hi!
could you state exactly in which mode and which firmware version and how you realized that you receive at +12kHz? Thanks in advance!

db4ple, 2016-07-13:
Hm. If you set -12khz, the base frequency (the one center the LO is tuning to) is Dial-Freq - 12khz.
This is how the code works. If you do the same with +12khz Center will be 12khz above dial freq.
And the VFO display matches that.

Regarding the red digits. I had a few of these recently. In the last commit I reversed a change we did when fighting the TwinPeaks, which effectively killed the delay handling in certain situations. Since then I did not have any red digits anymore.
I am now running the waterfall with max speed all the time, which puts a lot of stress on the machine and it responds nicely.

db4ple, 2016-07-13:
BTW, did you notice: The very latest firmware now has "System Information" instead of "Hardware Information" and you can check build time, firmware version etc. in this menu :-)

DD4WH, 2016-07-13:
did just notice now, it´s extremely useful! Great work, Danilo!

DG9BFC, 2016-07-13:
ok ... if i have a signal tuned to one vfo setting and in menue i have selected -12 kc ... then all is ok
when i change to + 12 kc the signal is gone
in older fw it worked fine (you could use high or low side if frequency and in all cases it worked and also vfo did match)

as an example i hear radio darc at 6070 (with menue set to -12 kc)

switch to +12kc and signal is lost

display jumps to low side (spectrum changes) but the receiving window is still on the high side

to receive again i have to turn the dial to 6056 (24 kc lower)

....

so it seems that the if shift calculation is somewhat "buggy" ... grin

it worked fine in older version (was it yesterday?? or day before yesterday??)

....

DG9BFC, 2016-07-13:
about red digits
a few versions before i could turn vfo VERY fast and the mchf was following with almost no problems
(just some hick-ups if si was to slow or had to do to big steps)
now the dial change stops after some turning ... even with smallest stepping (1hz) and no fast tune engaged (no blue box)
if i slowly and constantly turn ... after a few turns i have half a second stopping then change of numbers ... and again a short "pause" and so on
a few versions before the vfo was almost like an analogue radio (constant change and fast reaction)
but i can not exactly remember when that changed ( during last weeks i had very low spare time caused by new qrl/job)

(i maybe should have made two issues ... one foir the slowly vfo and one for the shift setting)

df8oe, 2016-07-13:
I have turned continously for more than 30 seconds - no red digits or stopping. There must be something that takes influence and that is different at your setting and my setting...

73
Andreas

db4ple, 2016-07-13:
Issue with +6/+12khz Translation is confirmed  after better testing. Thanks Sigi for spotting it.
Fix is coming.

db4ple, 2016-07-13:
Regarding the tuning issue (red digits). Since the timing was changed and we now favor the processing of encoders and keys over the spectrum display, it may be the case that this alters something in a way that tuning behaves less smooth (basically because we get much more encoder changes handled in the same time). Right now we have roughly 100hz change detection rate for encoders, we used to have something in between 40 to 15 per second.

db4ple, 2016-07-13:
We got sometimes I2C errors when tuning very fast over certain frequency ranges. I added a little more delay before retune and I was no longer able to get an I2C error from Si570.
Please report if the pull request #454 did not fix that. Even if it fixes the problem, we have look deeper why it happens in the first place.

db4ple, 2016-07-15:
Hi Sigi, please confirm that issues are gone in newer firmwares for you. I have the feeling that responsiveness of tuning has improved a lot by limiting the number of updates to about 25 per second.

DG9BFC, 2016-07-15:
ok ... if shift now is back to normal function :-) thanks on that
about the tuning issues
the response is ok .... but it still pauses while turning constantly ... THAT is the problem ... if i turn the vfo the mchf sometimes pauses the vfo change ... i am sure in an older version we had a lot faster tuning with no pauses ...
even when i turn SLOWLY ... (so just a few encoder pulses per second) there are these pauses
while the pause is (a fraction of a second) the spectrum is also freezed
i have to visit andreas and compare his settings with mine ... maybe then i find where our settings are different and what causes the cpu overload (it must be a short overload ... i have no other idea where these pause can come from)
....

df8oe, 2016-07-15:
The goal is that now polling of switches and encoders is INDEPENDENT of the system load...

DG9BFC, 2016-07-16:
tuning is fast and if shift works again ... so can be closed . (there are still pauses while turning vfo but i can live with it like it is now) ...

closed 2016-07-16 by DG9BFC