df8oe/UHSDR #459: band and mode change stops audio and restart codec does not solve that

DG9BFC, 2016-07-16:
sometimes when change from 12 to 10m (plus it switches then from usb to fm) the audio stops
restart codec does not bring audio back
same when change from 10 to 12m

so start in 12m usb and change to 10m fm.... silence (yes i have squelch open)
restart in 10m ... fm noise ... switch to 12m silence
restart in12m usb band noise ... switch to 10m silence ... and so on

tried resarting codec in both silence cases (see above) ... but no audio came back so resartet the whole rig and it worked again
changing modes on same band does not cause that fault .... but change mode plus band it stops receiving (no ..  receiving still works as can be seen in spectrum ... but no audio from speaker)

i then toggled trough bands till i got to radio darc on 6070 ... signal could be seen in spectrum ... but no audio from speaker
i did not use any noise reduction and noise blanker was set to 0 ...
fw used is 113 build juli 15 2016 22:53:27
(sometimes band change completely switched the unit off!!!!)

db4ple, 2016-07-16:
Hi Sigi,

this issue is for sure completely disconnected from the I2S - TwinPeaks issue.
Band change etc. have more or less nothing to do with the codec itself. 
Most likely we have some issue in the filter switching - fm demodulator part.
Once we can reproduce this issue consistently (which I think should not be too hard),
we  can start looking for the root cause. It may be a timing issue again.

DG9BFC, 2016-07-16:
here it seems a reproduceable fault
switching off and bacjk on solves it ....

DG9BFC, 2016-07-20:
tested newest daily snapshot (20.7. 18.35) ... still the same fault :-(

df8oe, 2016-07-20:
Not reproducable at all. Nobody else reports that issue. Maybe a hardware problem?

DD4WH, 2016-07-21:
Hi Sigi,

also tried that on my machine with your settings, but everything is
working, could not reproduce your issue . . . hmmm

Maybe  different settings in the menu, but I cannot think which ones should
be relevant here, so could possibly -as Andreas said- be a hardware issue .
. .

73 de Frank

DG9BFC, 2016-07-21:
hmmm i did not make any changes in hw ... and in older fw i did not notice that fault ...
seems as i really have to bring my mchf to andreas ... and then compare settings (and maybe try his settings in my machine?!?) ... it is a reproduceable fault here (as already said)

change from 10m fm to 12 m fm back and forth ... does work (for testing i also set 12m to fm)

then band change from 12 down to 15m stops audio

it is always when i habe a band plus mode change at once ...

change band when mode stqys the same and it works fine (for that test i set also 10m to usb)
when i have all bands set to usb/lsb then i can toggle trough the bands and it does not stop receiving
when any band is set to fm then after band and mode change the audio is lost

df8oe, 2016-07-21:
Switching in 12m to FM and then changing bands is like killing mcHF intentionally ;)

Do you have any problems if you *do not use FM at all*?

As told: Changing from FM may cause problems - this is not completely ready. We know that - but it is not urgent at all. If issue only occurs if you use FM I would close this issue...

DG9BFC, 2016-07-21:
hmmm sometimes i listen to a 10m fm repeater ... so i would like to have it working
i have a workaround ... i switch to ssb on 10m before change band ... and when back on 10m i then engange the fm mode ... seems to work as a temporarily solution ... so .... if it is the switching to fm mode ... and you are aware of that and it is already on the todo list ... yes then we can close this issue for now
but ... i know it worked fine in older fw versions and just wanted to point your nose on that fault ... so yes it can be closed for now (even if not solved)

closed 2016-07-21 by DG9BFC

df8oe, 2016-07-21:
It may work or it may not work. problems are located in code which interfers with other things (which almost nothing have to do with this code part). It was running by chance (and can run by chance in future without having done anything on this code part). Because this is a very, very rare usage its priority is very low. But it will be solved at next opening period of 10m of course :)

db4ple, 2016-07-22:
Hi Sigi,
I now found the time to reproduce the issue. I found it very easy to reproduce, good documentation of the issue from your side! I pushed a fix for the is
e issue in the pull request #450.
Basically a race condition caused the crash when the relay switching took place.
The old code switch the demod  mode indicator first, and only later switch the filters. In between that relay switching  takes places. It was long enough timewise that the audio interrupt came and crashed since some filter data structures were not set for the changed demod mode (FM -> SSB) . 
Please test, but for me the new code works (audio stays on), whereas the old code switch the mcHF off the moment I switched bands.

DG9BFC, 2016-07-22:
yes ... PROBLEM SOLVED  :-)
superb work
old version switched audio off (sometimes it completely switched mchf off!!)
new version is fine ... stays on in all cases (as far as i could test in a few minutes)
:-)