df8oe/UHSDR #465: Tx IQ adjustment does not work.

PA3CCE, 2016-07-24:
[Tx Signal Generation.txt](https://github.com/df8oe/mchf-github/files/380350/Tx.Signal.Generation.txt)

db4ple, 2016-07-24:
Hi  Jos,

thanks for submitting this as an issue.

For now I can only answer the last question: the tune mode does not switch while tune is active.

Danilo

DD4WH, 2016-07-24:
Hi Jos, hi Danilo,
yes, I can confirm the functioning of the single tone AND double tone TUNING. But you have to abort TUNE, adjust the tone (single or double) and then press TUNE again. That works perfectly with the newest version (I used the compiled version in the github at 17:00h) on my machine.

Will have a look at the other points.

73 Frank

df8oe, 2016-07-24:
Hi Jos,

I have just tested and TX IQ adjustment works fine using latest daily build.

Test-conditions:
mcHF xlate -12 KHz
testreceiver receiving at 14.0745 MHz
mcHF set to 14.000 MHz
transmitting in USB using TUNE function

I am able to reduce energy of unwanted signal so that it disappears in the noise floor of the test receiver (that is > 50dB).

73
Andreas

DD4WH, 2016-07-24:
Hi Jos, Andreas, Danilo,

TX IQ adjustment works perfectly, unwanted signal vanishes into the noise here.
However, of course other unwanted signals (spurios signals at) are still there, beware not to confuse those with the mirror rejection! 

73 de Frank

df8oe, 2016-07-24:
so it is...

73
Andreas

DD4WH, 2016-07-24:
Hi Jos,

coincidently, I made some similar measurements a few days ago on my mcHF. I measured after the PA, so not really comparable, but the figures seem to be on the same level, overall not really perfect, but acceptable (if you stay at 5 Watts! Do not mention Full Pwr, thats a totally different story --> you observed that also at full PWR adjustments).

Red Freq. display and non-working phase adjustment could be caused by Si570 solder connection problems?

The suppression at Fc-12k seems to be superb in your rig, the suppression at Fc-24k is unacceptable, but as we stated, this seems to be an issue in your system (or software version . . . ) as Andreas and I can adjust this mirror rejection value to very low values. The bandpass filter seems to suppress 2xFc by 20dB, good!
3xFc should be handled by the lowpass filter as usual, so no worries here.

Hope that helps!

Thanks a lot for your accurate measurements ! That makes me much more confident of my own measurements and makes me think that we can be safe to use the mcHF up to 5 Watts and comply to the regulations. 

All the best, 73 de Frank

PA3CCE, 2016-07-24:
Hi Andreas, Danillo and Frank,

Thank you all for the quick reactions.

OK, understood about the Two Tone switching, will do these measurements 
later on.

About the red frequency- display, I'm sure it has nothing to do with bad 
soldering of the Si570.
It happens the last few weeks much more than before, it worked fine 
then,  never happened.
It occurs very often when I switch from Tx to Rx, then the 
spectrum-display starts having
hick-ups and then, when I change the frequency or switch bands, the 
display turns red.
I have to switch off and on to have it running OK again.

I'm surprised that you all have the Tx IQ adjustment running.
Nothing happens here when I turn the encoders.
I just see some slight improvement on the Fc-72 kHz spurious when I turn 
the amplitude encoder,
but first I have to turn up the full power adjustment to overdrive to 
see that signal.
I should see reaction on the Fc-24 kHz signal I  suppose, but it stays 
steady.
Very strange that all seems to work fine apart from this, I will have to 
investigate it some more.

Will let you know, when I have more news.
best 73 de
Jos

df8oe, 2016-07-24:
Suppression using iq-adjust is working like a charm at -24kc. I can suppress the unwanted signal so well that it disappears :)To get a first working adjustment please take a look if iq adjustment works in rx. If so, please adjust tx values as rx values to get a good starting point. Using this values do have significantly suppression (not optimal but nearly optimal position). If gain OR balance is completely out of range you will *not find* a "resonance behaviour"!  So it is a good idea to adjust both simultaneously...

73
Andreas

PA3CCE, 2016-07-24:
Thanks Andreas.

Sure the Rx IQ adjustment always was working fine, I also could turn the 
unwanted signal down completely into the noise, which was
in the -110dB order. I will check if this is still the case.
The QSD and QSE devices are completely different and there sources 
likewise, how can it be that they require the same order of alignment ?
I do not understand that.

73
Jos

df8oe, 2016-07-24:
We have long worked on this to get rid of completely different settings. That was the first step to reach uniformed adjustments. I am nearly sure the differences between rx and tx now can be calculated so you only have to adjust rx OR tx in future. Differences in usb and lsb will hopefully be calculatable, too. AM and FM settings are quite another story at this moment (but we will work on it and I think it can be calculated, too). So the only big work must be to create  frequency dependent adjustments. I have tested but there is NO calculation possible :(

73
Andreas

PA3CCE, 2016-07-24:
Well, I have been working quite some time to reduce the frequency 
dependency of the Rx IQ adjustment.
It ended up in adding a ground loop of 1mm silvered wire around each 
function (LO , QSD- and QSE circuitry)
where all groundings of that circuitry were connected to very short.
Also added extra capacitors directly from the centertaps of the 
transformers in the QSD and QSE.
Also tried some different transformers there, but that appeared to make 
little difference.
It may be luck, but finally I can live now with just one setting over 
the entire frequency range !

73
Jos

df8oe, 2016-07-24:
That *is* luck (or related to your added modifications). I have made measurements now at more than 40 different mcHFs and there is a quite wide range iq<>working-frequency. But the ifferences rx/tx and usb/lsb are mathematically figured out...

We are working on this and hope we can offer a first solution the next three months.

73
Andreas

PA3CCE, 2016-07-24:
Well, I'm anxious to see that !!
vy 73

Jos

df8oe, 2016-07-25:
This is no issue - so I will close it.

closed 2016-07-25 by df8oe

PA3CCE, 2016-07-25:
Hi Folks

I found the cause :  low voltage over sampling capacitor C85.
Just have to find out why.

73
Jos