df8oe/UHSDR #468: Red Frequency display

PA3CCE, 2016-07-25:
[Red Frequency display.txt](https://github.com/df8oe/mchf-github/files/381938/Red.Frequency.display.txt)

db4ple, 2016-07-25:
Hi Jos,

does the same happen if you do not run tune for very long? I.e. is the long tuning the important thing to do? 
From the software point of view, tuning for more than a few milliseconds makes no difference. After around 50ms all relevant "long" term timers have expired or being rewound.
That means if it does not happen after 1s but after 1min most likely heat is causing trouble.
The only reason red freq display is shown is some sort of I2C error, i.e. the SI570 does not respond as expected to a normally working command. There are two reason we identified: Too fast switching of frequencies. Have been working on preventing that in the recent firmwares but seems to be working since about 2 weeks or so, see the issue #459, where this was also part of the issues mentioned.

The other reason is a intermittent error in the Si570 communication often happening due soldering issues. And these could be happening due to heat as you know.

To verify: Get a red display, wait a bit to cool down the mcHF and try moving the dial. This should eliminate the red display since we retry the I2C communication with every dial move. If this eventually brings back the white numbers, check the soldering. If not, I am willing to think about how to more clearly identify the issue with the help of software modifications. 

Meanwhile I will get my dummy load out and heat it a bit. Maybe I am able to reproduce but honestly, I doubt that, since I have been working on that very recently and check this very scenario (band switching causes red display). I could reproduce the issue before but not after fixing the found issue (timing related).

73
Danilo

PA3CCE, 2016-07-25:
Hi Danillo,

I have a strong feeling it has to do with the tune function, for when I 
stay in Rx mode I can tune around with small and big steps, fast and 
slowly, switch bands, all does fine.
As soon as I have been in tune mode things go wrong and it looks like it 
has  to do with relatively short or long periods of tune.
As you know I do not have the PA fets mounted, so temp differences in Rx 
and Tx modes are still small.
The voltage regulators are mounted on a separate cooling strip; This 
strip gets hot but is not close to the Si570.
The PCB's are interconnected by a homebrewed interconnection board, they 
are in free air so I don't expect temperature will be an issue.
Sometimes it also happens that the tune button does not react after 
having been in tune (not often).
When I switch off/on the display is in white  immediately and I can tune 
around as long as I stay in Rx mode.
Will go on trying to find out what is  the combination that causes it.

Regards
Jos

db4ple, 2016-07-25:
Hi Jos,

hm. yes, Temperature may not be the cause given your setup. Please do me one favour, though.
Instead of using the tune function (which you're running in SSB, right?), please press PTT for an equivalent amount of time required to getting the red digit with tune. Please also let me know what is long tune in seconds, so that I have a way to reproduce the scenario.

Thanks,
Danilo

PA3CCE, 2016-07-25:
>
> Hi Jos,
>
> hm. yes, Temperature may not be the cause given your setup. Please do 
> me one favour, though.
> Instead of using the tune function (which you're running in SSB, 
> right?), please press PTT for an equivalent amount of time required to 
> getting the red digit with tune.
Well , would be pleased to do so, but I am not yet equipped to do a push 
button test.
I'm still in testing mode you know, so I did not make plug + cable and 
so on.
I could make a short on the board but ...
>
> Please also let me know what is long tune in seconds, so that I have a 
> way to reproduce the scenario.
Well, that's difficult to say, I was doing the tests while it all 
happened, did not record the time,
but it was the time needed to do all test per frequency line , let's say 
10 to 15 minutes, maybe even more.
I can't say if length of time is really important, but I can't reproduce 
myself, when I do it quickly
This made the time of measuring all of it even more lengthy, that was 
annoying.
Now I have to do it all over again, after repairing the IQ correction. ;-)

df8oe, 2016-07-25:
The maximum time I can stay in tune mode is 5 minutes - after that time PA idle would blow up my PA. But 5 minutes seem not be long enough - I cannot reproduce here :/

73
Andreas

db4ple, 2016-07-25:
Well,
since it does not have a power related factor (without finals I doubt that the power itself is relevant for that test), I am running now with 0.5 W and will wait for at least 15min.

df8oe, 2016-07-25:
Because of cooling of PA is here only the metal case idle without any output is strong enough to heat up the case > 60°C. So pushing PTT in SSB without audio is not possible for more than 5 minutes, too... Intermittent usage with transmitting < 3min (which often occurs in normal operation) never causes a red digits issue. Can it be related to temperature compensation routine ? No idea...

73
Andreas

PA3CCE, 2016-07-25:
>
> The maximum time I can stay in tune mode is 5 minutes - after that 
> time PA idle would blow up my PA. But 5 minutes seem not be long 
> enough - I cannot reproduce here :/
I see, but I really don't think it must be that long.
I also have difficulty in reproducing it while I'm not measuring, but 
succeeded some times with a short tune of a few seconds.
As said, while doing the measurements it happened very often.

df8oe, 2016-07-25:
May be you are coupling rf to lines where no rf should be? Drivers do have up to 20Vss at their collectors, and if you attach a probe it can get pulses that  disturbes I2C...

Would be interesting if the issue occurs without touching any rf lines during TX.

73
Andreas

db4ple, 2016-07-25:
3 more minutes :-(

db4ple, 2016-07-25:
Hi,

you will not believe it: 15 min, 0.5W and I was able to crash the mcHF somewhat. The temperature display showed crazy random values, sometimes --.- sometimes 8.1 or so, sometimes 96.5 etc.
I was not able to get the mcHF to respond anymore.
Unfortunately I was not connected  via ST-Link, so I could not find out what was going on inside the software. Absolutely strange.

Will let the thing cool down a bit and run  it with ST-Link this time. We may be onto something.

73
Danilo

PA3CCE, 2016-07-25:
>
> May be you are coupling rf to lines where no rf should be? Drivers do 
> have up to 20Vss at their collectors, and if you attach a probe it can 
> get pulses that disturbes I2C...
>
> Would be interesting if the issue occurs without touching any rf lines 
> during TX.
Well, I don't think so.
I have the scope connected to the PCB with  probes (set for divide by 
10) plugged directly into the holes for the bases of the mosfets.
Further more I have toroids in the cables to the input of the scope, to 
prevent for ground currents.

PA3CCE, 2016-07-25:
Please be careful not to destroy the mcHF.
I hardly can believe that it's the temperature here.

Jos

PA3CCE, 2016-07-25:
The probes are connected to gates of mosfets ofcourse ;-):-P

DD4WH, 2016-07-25:
Hi folks,
I had the same fault yesterday, when I did my measurement session on the TX filters! Very similar condition: long periods of transmitting and wild switching and adjusting around in the menu while TUNE, pressing PTT etc.
My first thought was a short circuit or something (very recently built in a switching power supply, so I thought it was that!), because the temperature display was at zero degrees . . .

Occured only once in about three hours of continous running of the mcHF with low power transmit into a dummy load. Same reaction: red digits, very slow response while tuning, only went away after restarting the mcHF.
I saw this phenomenon only once.
73 de Frank

db4ple, 2016-07-25:
Hi,

okay, I think we may have found something: I was watching the TCXO temp rising towards 52 degrees C
and suddenly the temp measurements went crazy. A little difficult to debug but it seems it then got very wrong readings from the temp sensor. 
Need to investigate more but I have no

PA3CCE, 2016-07-25:
Hi Frank, welcome

I just could produce strange behavior several times in a row :

Switch on (mcHF was set for 5.4000kHz
Tune for 5 seconds.
Switch off Tune.,
Change bands.
Press Tune again (tune goes gray)
Press Tune again(tune goes white)
Press Tune again (tune remains white)
Have to switch mcHF OFF/On to go back to normal

Jos

PA3CCE, 2016-07-25:
OK, but to my humble opinion this is not related to the red display problem.

Jos

DD4WH, 2016-07-25:
yes, long press TUNE disables TX and "TUNE" is displayed in grey, that´s a feature ;-)

PA3CCE, 2016-07-25:
OK, but I did not press Tune long.
Jos

PA3CCE, 2016-07-25:
Did I say that ?
No I pressed tune and then waited 5 seconds

Jos

db4ple, 2016-07-25:
Hi Jos,

you're that the new band was not in AM/FM/SAM? In this case tune does not switch to tune mode.

Anyway, will now check the temperature lead.

PA3CCE, 2016-07-25:
Hi Andreas

No I'm sure it wasn't, it still is in LSB mode.
At the moment I cannot reproduce it anymore.
I'm wondering, could it be related to bouncing switches ?
Anyhow, I think I call it a day for now, have been busy all day, time to 
relax.

Have a good night
Jos

PA3CCE, 2016-07-27:
Hi Andreas, Danillo, Frank

Please find attached the updated Tx test, amazing !

73 de
Jos

mcHF Tx SIGNAL GENERATION UP TO AND INCLUDING THE PA DRIVER.
============================================================
Updated 26/07/2016

Conditions :
++++++++++++

1) The latest daily Firmware.
2) mcHF in "Tune" mode, "LSB", "Rx/Tx Freq Xlate" = "Rx-12kHz" , "Tune Tone (SSB)" = "single". 
3) PA mosfets Not mounted.
4) Measuring Instrument : Digital Scope/Spectrum Analyzer  Picoscope 3207B.
5) UI board version 0.4
6) RF board version 0.5
7) Potatoechips Si570/4 divider.
8) All IZ6MAF mods implemented.

DC Measuemnts :
+++++++++++++++
DC in 	12.03V
+8V	 8.47V	
+5V	 4.94V
+3V	 3.28V

QSE BIAS  1.52V

Driver Q3,Q4, 2x DXT3150
------------------------
Vcc_12V 11.42Vdc
PTT_5V 	 4.76Vdc

Q3E	 1.79Vdc	Ic = 38mA
Q3B	 1.32Vdc
Q3C	11.45Vdc

Q4E	 1.80Vdc	Ic = 38mA
Q4B	 1.32Vdc
Q4C	11.44Vdc

AC  Measurements :
++++++++++++++++++

1) TLV2464 In : (11.250 kHz) :
++++++++++++++++++++++++++++++
Full PWR Adjust = 30

In I :	 287 mVrms
In Q :	 269 mVrms

2) TLV2464 Out : (11.250 kHz) :
+++++++++++++++++++++++++++++++
Full PWR Adjust = 30

Out1 :	 286 mVrms
Out4 :	 286 mVrms
Out3 :	 269 mVrms
Out2 :	 268 mVrms

3) LM386 Out : (11.250 kHz) :
+++++++++++++++++++++++++++++
Full PWR Adjust = 30

Output LM386's, adjusted by the input attenuators.

U20 :	785 mVrms
U21 :	787 Vrms
U22 :	765 Vrms
U23 :	765 Vrms

4) Mixer-In voltage test @ R63 (11.250 kHz):
++++++++++++++++++++++++++++++++++++++++++++++

Full PWR Adjust   Level Vrms	Delta (mV)		 	
----------------  ----------	----------		
     10		     0.262	   0   			
     20		     0.524	  262  
     30		     0.786	  262  
     40		     1.048	  262 		
     50		     1.309	  261 
     60		     1.565	  256 
     70		     1.827	  262 	
     80		     2.089	  262  	
     85(max)	     2.218	  127	

5) Tx mixer out over 51E (LSB)
++++++++++++++++++++++++++++++

Full PWR Adjust = 30
    
		Power(Fc)	Fc-12 kHz	Fc-24kHz	   2Fc		  3Fc	
		---------	---------	---------	 --------	---------	
 3.6 MHz 	 3.7 dBm	-53.5 dBc	-48.5 dBc	-50.6 dBc	-10.3 dBc	
 5.4 MHz  	 3.1 dBm	-50.3 dBc	-50.2 dBc	-32.5 dBc	- 9.2 dBc		
 7.1 MHz  	 3.7 dBm	-48.5 dBc	-52.3 dBc	-46.6 dBc	- 8.7 dBc							
10.1 MHz  	 3.4 dBm	-45.1 dBc	-55.7 dBc	-43.5 dBc	- 8.7 dBc	
14.2 MHz  	 3.8 dBm	-42.2 dBc     < -60.0 dBc	-40.5 dBc	- 8.2 dBc	
18.1 MHz  	 3.5 dBm	-40.9 dBc     < -60.0 dBc	-37.6 dBc	- 5.8 dBc
21.3 MHz  	 3.9 dBm	-41.0 dBc	-53.0 dBc	-35.2 dBc	- 6.7 dBc
24.9 MHz  	 3.9 dBm	-37.2 dBc	-49.8 dBc	-33.0 dBc	- 8.8 dBc
28.5 MHz  	 3.9 dBm	-35.3 dBc	-46.3 dBc	-32.0 dBc	- 8.8 dBc

6) Tx Bandfilter out over 51E (LSB)
+++++++++++++++++++++++++++++++++++
Full PWR Adjust = 30

		Power(Fc)	Fc-12 kHz	Fc-24kHz	   2Fc		  3Fc	
		---------	---------	--------	 --------	---------	
 3.6 MHz 	 1.8 dBm	-52.7 dBc	-48.6 dBc	-54.2 dBc	-34.6 dBc	
 5.4 MHz  	-2.0 dBm	-49.5 dBc	-50.3 dBc	-56.6 dBc	-32.2 dBc		
 7.1 MHz  	 1.2 dBm	-49.6 dBc	-51.4 dBc	-53.9 dBc	-36.8 dBc							
10.1 MHz        -1.6 dBm	-45.7 dBc	-54.8 dBc	-48.0 dBc	-25.3 dBc	
14.2 MHz  	 1.0 dBm	-43.1 dBc	-63.9 dBc	-54.1 dBc	-29.2 dBc	
18.1 MHz  	 1.7 dBm	-39.6 dBc	-60.9 dBc	-41.0 dBc	-21.0 dBc
21.3 MHz  	 2.5 dBm	-40.6 dBc	-53.9 dBc	-44.6 dBc	-23.4 dBc
24.9 MHz  	 2.3 dBm	-37.9 dBc	-53.7 dBc	-47.1 dBc	-25.5 dBc
28.5 MHz  	 2.1 dBm	-36.0 dBc	-48.5 dBc	-47.6 dBc	-26.5 dBc

7) PA Driver out over R81 = 220E :
++++++++++++++++++++++++++++++++++
Full PWR Adjust = 30			IQ adjustment
								Fc-24kHz	Fc-24kHz					       
		Power(Fc)	LSB Tx Bal.	LSB Tx Phase	adjusted       106/-32 set   	Fc-12 kHz	   2Fc		  3Fc			
		---------	-----------	------------	--------       -----------	---------	---------	---------  		
 3.6 MHz 	25.72 dBm	     96		    -32		< -60 dBc	-50.0 dBc	-50.0 dBc	-26.1 dBc	-40.8 dBc			
 5.4 MHz  	23.56 dBm	     97		    -32		< -60 dBc	-49.8 dBc	-50.7 dBc	-25.0 dBc	-38.0 dBc	    	  
 7.1 MHz  	22.44 dBm	     99	    	    -31		< -60 dBc	-52.4 dBc	-46.0 dBc	-24.8 dBc	-37.9 dBc							
10.1 MHz        18.92 dBm	    101	    	    -30 	< -60 dBc	-54.9 dBc	-47.6 dBc	-28.7 dBc	-34.6 dBc		
14.2 MHz  	21.17 dBm	    104	            -32		< -60 dBc	-63.3 dBc	-44.2 dBc	-26.5 dBc	-38.3 dBc   		
18.1 MHz  	19.72 dBm	    108	    	    -30		< -60 dBc	-61.0 dBc	-38.8 dBc	-28.2 dBc	-32.4 dBc 	
21.3 MHz  	19.21 dBm	    112	    	    -34		< -60 dBc	-53.1 dBc	-40.6 dBc	-27.9 dBc	-35.7 dBc   
24.9 MHz  	17.75 dBm	    113	    	    -34		< -60 dBc	-51.8 dBc	-39.6 dBc	-26.2 dBc	-38.4 dBc	
28.5 MHz	16.40 dBm	    116	    	    -36		< -60 dBc	-49.2 dBc	-37.1 dBc	-25.0 dBc	-40.6 dBc    	

8) PA Driver out over R82 = 220E :
++++++++++++++++++++++++++++++++++
Full PWR Adjust = 30
										Fc-24kHz
		Power(Fc)						       106/-32 set	Fc-12 kHz	   2Fc		  3Fc	
		---------							---------	--------	 --------	---------	
 3.6 MHz 	25.69 dBm							-50.6 dBc	-50.6 dBc	-25.8 dBc	-39.9 dBc	
 5.4 MHz  	23.54 dBm							-50.8 dBc	-50.7 dBc	-24.9 dBc	-37.8 dBc		
 7.1 MHz  	23.19 dBm							-53.2 dBc	-48.0 dBc	-24.7 dBc	-37.9 dBc							
10.1 MHz        18.70 dBm							-54.7 dBc	-47.4 dBc	-30.8 dBc	-35.9 dBc	
14.2 MHz  	21.32 dBm							< -60 dBc	-34.2 dBc	-26.2 dBc	-37.9 dBc	
18.1 MHz  	19.88 dBm							-57.3 dBc	-39.8 dBc	-26.9 dBc	-33.2 dBc
21.3 MHz  	19.54 dBm							-51.9 dBc	-40.3 dBc	-27.0 dBc	-34.4 dBc
24.9 MHz  	18.15 dBm							-50.6 dBc	-39.2 dBc 	-25.5 dBc	-36.7 dBc
28.5 MHz  	16.85 dBm							-48.6 dBc	-36.7 dBc	-24.5 dBc	-39.1 dBc

Remarks
+++++++

These just are my findings on my specific mcHF.

All results are obtained by setting the LSB IQ settings in the menue to 106 (Bal.) and -32 (Phase)
according to the findings in 7).
 
Full PWR adjustments more than 40 is causing increased spurious closely around F-carrier.

I suffered from red Frequency display while switching from Tx to Rx and to different bands
many many times. Had to restart the mcHF often to solve this. 

To Do :
-------
Measurements with "Two tone" signal.

Best 73 de 
Jos PA3CCE

DD4WH, 2016-07-27:
Hi Jos,

thanks a lot for your valuable measurements. I have not yet had the time to go into full detail.

However, before you carry out other measurements in double tone or single tone OR SSB, please wait a few days (maybe weeks!??). We have identified a software problem in the production of the TUNE tone, which could potentially blur all your measurements. Additionally, I am preparing (and have already implemented a part of it) a review of the TX path with other filters to make the signal cleaner. So before you carry out more measurements, please wait for the fix of that problem and the TX filter review. It is not trivial, so could take us some time. Thanks!

73 de Frank

PA3CCE, 2016-07-27:
Hi Frank,

Yes I read about the issue.
I tried 2 Tone, using yesterdays firmware, see attachment.
If you are interested in some more of these, please let me know.

73
Jos

df8oe, 2016-07-27:
Hello Jos,

very interesting and detailed measurements.... But it would be much more interesting to do this with reliable signal quality in "tune" - what we do not have at the moment... Investigations about the code written by Clint were started but it is *not a simple task*. So, as Frank told you, this may take some time. I don't think you should spend more time NOW in measurements - the results can differ widely when DDS-Code is fixed.

Regarding the "red digits issue" I took the mcHF which is build as an "RX only" (no PA, drivers and tx-mixer installed) this morning. "I transmitted" many, many times for 2...15 minutes but never I saw red digits when switching back, changing bands or doing anything else. I am not able to reproduce this here.

Please can you try:
1) switch TCXO to "OFF"
2) dont touch hardware with probes during tx phase

Does one of the above (or both together) affect red digits at your mcHF?

73
Andreas

PA3CCE, 2016-07-27:
> Please can you try:
> 1) switch TCXO to "OFF" I will see what happens
>
> 2) dont touch hardware with probes during tx phase. Never did
>
> Does one of the above (or both together) affect red digits at your mcHF?
>
> 73
> Andreas

> Hello Jos,
>
> very interesting and detailed measurements.... But it would be much
> more interesting to do this with reliable signal quality in "tune" -
> what we do not have at the moment... [...]
During testing for the update, I still had it, even in simple sequences.
I keep having the feeling it has to do with some waiting time in Tx

df8oe, 2016-07-27:
I have waited 2...15 minutes - without result.

But without modifications I cannot test this at another mcHF because all of them gets overheated when they transmit > 3 min. They do have nice housings and no cooling other than the housing itself...

3
Andreas

PA3CCE, 2016-07-27:
>
> I have waited 2...15 minutes - without result.
I don't think more than 2 minutes is necessary, even less.

df8oe, 2016-07-27:
Many OMs are transmitting everyday many times 1...3 minutes "as one part" - I do this very often. No other report about red digits, I never have had seen it by myself. I am nearly sure it is a hardware issue at your device...

73
Andreas

PA3CCE, 2016-07-27:
While switching TCXO off I found a bug there.
When setting the options you can set to "ERROR" (b.t.w. wat does this mean?)
When setting to "On" or "OFF" the E from ERROR stays there.

PA3CCE, 2016-07-27:
>
> Many OMs are transmitting everyday many times 1...3 minutes "as one 
> part" - I do this very often. No other report about red digits, I 
> never have had seen it by myself. I am nearly sure it is a hardware 
> issue at your device...
Can't believe that

PA3CCE, 2016-07-27:
>
> Many OMs are transmitting everyday many times 1...3 minutes "as one 
> part" - I do this very often. No other report about red digits, I 
> never have had seen it by myself. I am nearly sure it is a hardware 
> issue at your device...
Just bouncing  could be an issue

df8oe, 2016-07-27:
Manythanks for this issue report. I fixed the menu display in the latest daily build. But there was no functional error at all - only the displayed text.

Please have a look at Yahoo groups or ur German discussion group - nothing related to "red digits". And transmissions 1...3 minutes are absolutely normal if you work in SSB.

What do you mean with "bouncing"? I can toggle PTT dozens of times, can tx for 1 second or for 3 minutes - no red digits at all.

73
Andreas

PA3CCE, 2016-07-27:
> Manythanks for this issue report. I fixed the menu display in the
> latest daily build. But there was no functional error at all - only
> the displayed text.
>
> Please have a look at Yahoo groups or ur [...]
Bouncing is when you press a button, the contact opens and closes more 
than one time.
However, I have a strong feeling I found the cause of the red figures.
When you press Tune long, the Tx mode gets blocked and Tune turns gray.
When you tip the button all is fine.
When you press the button distinctly, something goes wrong, Tune remains 
white and the spectrum display starts having "Hick ups",
then when you switch bands figures go red !.
Just a matter of programming a bit longer delay to determine if it's 
long or short press.
When I'm wrong, sorry.
Jos

Andreas

db4ple, 2016-07-27:
Hi Jos,

if you could produce a small video showing what you just described, it would be very helpful. You can attach the video file right to the message, like you did with the txt files.

Thank you!

PA3CCE, 2016-07-27:
>
> Hi Jos,
>
> if you could produce a small video showing what you just described, it 
> would be very helpful. You can attach the video file right to the 
> message, like you did with the txt files.
Hi Danillo

Sorry, I have no video camera available at this moment.
Perhaps you can see in the source code what might go wrong ?
I could reproduce this a few times.
Fact is that Tune stays white but other things seem to happen.
Might also be I'm completely wrong :-[

Jos

df8oe, 2016-07-27:
I have tested TUNE-Button regarding function "tune enable", "tune disable", "tx disable" and "tx enable". Everything works as expected here. So I cannot see anything in the code what canreult in an issue I cannot reproduce nor I can understand what I must do to get it. I think we wait until you can send a video (or somone other reports same)...

73
Andreas

PA3CCE, 2016-07-27:
> I have tested TUNE-Button regarding function "tune enable", "tune
> disable", "tx disable" and "tx enable". Everything works as expected
> here. So I cannot see anything in the code what canreult in [...]
OK, the difference was just a short tip or a distinct (so somewhat 
longer press), then it happened a few times with me.
Problem is, what is short and what is just long enough to make it happen.
Sorry, was just a thought.

DD4WH, 2016-07-27:
Hi Jos,
I tested the function of the TUNE button in my mcHF with long presses, short presses, very short presses, many presses following each other in very short time etc. Also tested to change band between presses.

Nothing unusual here. Everything seems to work fine. I also noted that you need to be patient sometimes, because the TUNE button needs some 500ms or so to settle until it reacts again after heavy treatment. But even then, no red digits or unexpected behaviour.

Please be more explicit and precise in the description of the behaviour of your mcHF. It is quite difficult to understand what you mean:
##Fact is that Tune stays white but other things seem to happen.##
Stays white when you press TUNE? Which other things, how do you know?

##When you press Tune long, the Tx mode gets blocked and Tune turns gray.##
That´s how it should be-->normal behaviour

## When you tip the button all is fine. When you press the button distinctly, something goes wrong##
What is "tip", what is "press distinctly"? Please be more precise, so we can interpret what you are doing.

An even better alternative, as Andreas and Danilo stated, would be a video!

Good luck! 73 de Frank

PA3CCE, 2016-07-27:
By the way, I do have the TCXO off (bullet is red), but it still happens.

Jos

PA3CCE, 2016-07-27:
> Hi Jos,
> I tested the function of the TUNE button in my mcHF with long presses,
> short presses, very short presses, many presses following each other
> in very short time etc. Also tested to change [...]
Sorry when I was not clear.
Yes, when I press tune to switch from Tx to Rx.
I don't know for sure, that's why I said "seem to happen" but The Tune 
word on the display stays  white.
>
> ##When you press Tune long, the Tx mode gets blocked and Tune turns 
> gray.##
> That´s how it should be-->normal behaviour
I know that it is normal behavior, I mentioned it to memorize that.
>
>
>     When you tip the button all is fine. When you press the button
>     distinctly, something goes wrong
>
> What is "tip", what is "press distinctly"? Please be more precise, so 
> we can interpret what you are doing.
How can I say that more clear ?
Tip is a very very short press (more like "OOPS") and distinctly a very 
clear so longer one.
Is this a language related misunderstanding ?

73 Frank

Jos

df8oe, 2016-07-27:
I can tip, press, press & hold and do all combinations in every state - but reactions are all as expected. Nothing happens beneath the wanted actions, and there are no red digits.

73
Andreas

PA3CCE, 2016-07-27:
>
> I can tip, press, press & hold and do all combinations in every state 
> - but reactions are all as expected. Nothing happens beneath the 
> wanted actions, and there are no red digits.
OK, thanks,  let's leave it for the moment.
I feel some irritation here and there, and that is not my intention.

73
Jos

df8oe, 2016-07-27:
I  know this because we do have contact before, Jos.. We will see in future. And very interesting will be the investigations about DDS tone that are running @ the moment ;)

73
Andreas

PA3CCE, 2016-07-28:
Just for fun, see attachment.

PA3CCE, 2016-07-28:
Some more fun

df8oe, 2016-07-28:
There is no attachment...

PA3CCE, 2016-07-28:
Hoow can it be ?
Saw it on my screen.
Did you receive the second one ?

db4ple, 2016-07-28:
Sorry, you need to attach the files directly in github Web interface, it will not take attachments from mails. Use "view it on Github"

df8oe, 2016-07-28:
...or post the links (url)...

PA3CCE, 2016-07-28:
OK, will do.
It worked with .txt files

db4ple, 2016-07-28:
Oh, did not knew that. You live to learn...

PA3CCE, 2016-07-28:
>
> Oh, did not knew that. You live to learn...
That's for sure.
Did send them via Github

df8oe, 2016-07-28:
Some of the spurs (many??!!) are the result of a "bad working-together-chain" DDS-->TX-filters-->Hilbert-filters.

We are "on a track" to solve this but because of holidaytime it will take weeks to understand & fix that. But now we are full of hope to get rid of this worse suppressions. Many spurs are only at TUNE functionnot in normaloperation mode. But is is without anysense to implement a two tone generator if its signal is unusable :) Be patient - we will work on this :)

df8oe, 2016-07-28:
I agree, Jos. "We can live with it" - it is a QRP transceiver and homebrew. But I think we can improve much... Surprise, surprise ;)

PA3CCE, 2016-07-28:
>
> I agree, Jos. "We can live with it" - it is a QRP transceiver and 
> homebrew. But I think we can improve much... Surprise, surprise ;)
Yes, and keep in mind that these results are just from the driver alone.
I think I will go install the mosfets now and see what happens then.
73 de
Jos

df8oe, 2016-08-02:
No "red digits issue" found / reported by others.

closed 2016-08-02 by df8oe