df8oe/UHSDR #476: Just for fun

PA3CCE, 2016-07-28:
[image]
[image]

DD4WH, 2016-07-28:
Your highest spur is about -50dBc. Thats not too bad, but we can do better. Please wait a few weeks, then hopefully we will have fixed. The latest preliminary version (not in github yet) has -57dBc and we would like to still improve it ;-).
73 de Frank

PA3CCE, 2016-07-28:
OK Frank, I found these results not too bad indeed.
I'm anxious to see the results of the inprovements, but don't hurry, 
take your time.
73  de Jos

DD4WH, 2016-07-28:
. . . and IMD3 of 37dBc PEP is also very good! What TX power was that?

PA3CCE, 2016-07-28:
>
> . . . and IMD3 of 37dBc PEP is also very good! What TX power was that?
Please remember, it was just @ driver out (one half) Full power setting 
@ 30 in the menu.
Mosfets not mounted yet.

PA3CCE, 2016-07-28:
>
> . . . and IMD3 of 37dBc PEP is also very good! What TX power was that?
Sofar I did my measurements no mosfets mounted.
Voltage over R81 @ 14.2 MHz single tone was 2.6V in this case.

df8oe, 2016-08-11:
Nice to see.  Not sooooo bad !!

But it will get better ;)

73
Andreas

closed 2016-08-11 by df8oe

PA3CCE, 2016-08-11:
>
> Nice to see. Not sooooo bad !!
Not bad at all, now I measured with PA included, vy reasonable results.
>
> But it will get better ;)
Hope so, I  am vy anxious to see.