df8oe/UHSDR #488: use 24L1025 bug

BI3MEK, 2016-08-06:
1.Use mchf_firm_9_219_26_4, EEPROM can be stored properly
Use 0.219.27.12,It display 24xx1025 / 128kb, red font, can not save settings
Use 1.0.0, display 24xx1025 / 128kb, red font, can not save settings
use 1.1.1, display 24xx1025 / 128kb [too small], red font, can not save settings
Use just repaired version [# 487] is displayed 24cm02 / 256k [error], red font, can not save your settings.

db4ple, 2016-08-06:
Hi,

thank you. Could you please tell which EEPROM causes the problems (is it this one: https://cloud.githubusercontent.com/assets/18715486/17455746/756a1d48-5bf4-11e6-8598-09811a05572d.jpg ?)

73
Danilo

BI3MEK, 2016-08-06:
yes,it is.

73
bi3mek

df8oe, 2016-08-06:
Please tell if problem comes up even after resetting EEPROM in configuration menu. If there is damaged data inside red digits and/or unusable state will survive even if firmware works correct. Detecting only is done on NEW AND UNUSED (or resetted) EEPROMS, detection process stores the result in EEPROM. If this step fails it is wrong FOREVER until EEPROM is resetted again.

73
Andreas

db4ple, 2016-08-06:
The problem here is, that first of all the detection seems to fail to detect the right chip.
Secondly, even if we assume it we can fix that to detect a 24xx1025, it seems that it does not work either, since in older firmware it was detected as 1025 but did not work either. 

The firmware mchf_firm_9_219_26_4 does not use an I2C EEPROM, it uses flash only, as far as I can see (was checking the source at https://github.com/m0nka/mchf-github ).
So we  can conclude: This EEPROM never worked in the current firmware. Maybe it is a problem  in  the EEPROM or the 24L1025 behaving not like the 24xx1025.

73
Danilo

BI3MEK, 2016-08-06:
I'll replace it with a new chip, and then report, thank you！

73
bi3mek

df8oe, 2016-08-06:
I have tested a couple of minutes before with:

24LC1025
24LC1026

works flawlessly. Because of EEPROMS were used before I resetted them using configuration menu. EEPROM was detected after this correctly and used with settings from backup in virtual EEPROM - all as expected.

I do not know 24L1025 (without 'c').

73
Andreas

db4ple, 2016-08-06:
Yes, the "there-is-an-answer-for-everything"-internet does not really tell us what a 24L1025 (may be just a marking style  of 24LC1025, maybe a 24LC1025 clone.

Lets see what the other EEPROM is doing.

BI3MEK, 2016-08-07:
Re-purchase a new chip package, a width of 5.2mm, printed characters 24LC1025. Here is the test

A. use 1.1.4 version
1. Turn on the display 24XX1025 / 128K [used], green.
2. reset Ser EEPROM,
3. reboot the mcHF, EEPROM display 24xx1025 / 128K [error], red
4. Press PWR button, shutdown, reboot, display 24XX1025 / 128K [used], green.

use 0.219.27.12 version
1. Turn on the display 24XX1025 / 128K, green
2.reset Ser EEPROM,
3.reboot the mcHF, EEPROM display 24xx1025 / 128K, red
4. Press PWR button, shutdown, reboot, display 24XX1025 / 128K, green.

Thank you very much, 24LC1025 can now be used as normal!
73
BI3MEK

BI3MEK, 2016-08-07:
[image]

closed 2016-08-07 by BI3MEK