df8oe/UHSDR #496: Frequency display issue when enable split function in cw mode(DF8OF firmware V1.0.0 to last daily snapshot version))

patiwatb, 2016-08-08:
Hello
Hello
I test cw mode with split function.I have the problem about display.
When tuning frequency VFO A , VFO B difference TX/RX then start CW keying the frequency display go back.(Current rx frequency not change but change only rx frequency display)
to the same frequency.This problem only cw mode.SSB mode no problem
I setting cw tx/rx delay 0 or any number have same problem
73 DE HS1PGS(Patiwat)

db4ple, 2016-08-09:
Hi Patiwat,

thanks for providing all necessary information. It was a kind of hard to find issue.
Please try the newest daily build. It should fix the issue. 

If the issue is fixed, please close the issue, if not, we'll see if there is more. Thank you.

73
Danilo

patiwatb, 2016-08-10:
Hi Danilo
Thank for the fast respond to fix problem.
I will upgrade firmware and ask my friend in Thailand to test also tonight.Then I will update issue and confirm again.
73 DE HS1PGS(Patiwat)

patiwatb, 2016-08-10:
Hi Danilo
I confirmed with my Thailand MCHF group.
Last firmware can solve the problem.
Thank again to resolve very fast.
73 DE HS1PGS(Patiwat)

closed 2016-08-10 by patiwatb