df8oe/UHSDR #544: cannot select menue item

pointhi, 2016-08-19:
when I'm inside the menue, I should be able to select a menue item using ```M2``` (when reading the manual correctly). But no button at all is able to select the menue entries. Switching to another menue-item using ```ENC2``` for example is not a problem. The Button works, because in normale mode I'm able to change options on the display using ```M2``` so it seems to be a firmware bug.

I'm using a ```mcHF v0.5``` with an up-to-date firmware.

73, OE5TPO

DD4WH, 2016-08-19:
Hi!
Use encoder 3 to select entries in the menu.
I do not know which manual you read, but there has -to my knowledge- never been a possibility to select menu entries by buttons.
If this solves your problem, please close this issue.
73 de Frank

pointhi, 2016-08-19:
thx. I thought I tried every knob on the device.

closed 2016-08-19 by pointhi