df8oe/UHSDR #557: CW issues mchf.bin 1.2

steinhj, 2016-08-31:
1. Looks like selection of iambic_a is broken in mchf.bin 1.2. 
    There is no difference between IAM_A and IAM_B. They both act as IAM_B.

2. Automatic ModeSelection depending on band ( CW-U or CW-L) does not work in mchf.bin 1.2.
    Looks like it is doing the opposite choice and selecting CW-U on the lower bands and 
    CW-L on the upper bands

df8oe, 2016-09-01:
Please post your CW Freq. Offset. These settings are confusing to most of the users..

df8oe, 2016-09-01:
I hope 2) is fixed now in actual daily snapshot. Already working on 1) - please be patient.

df8oe, 2016-09-01:
Can you please tell in which version it was working correct in IAMBIC A? I don't see any significant functionally changes in this code part since code released by KA7OEI! Because of I was working always in IAMBIC B I never have noticed changes related to IAMBIC A...

steinhj, 2016-09-01:
Thanks for looking in to this so fast, and for the great job you are doing.
My last installed version was 1.0, and the iambic_A was working in this version.

Stein, LA2FDA

df8oe, 2016-09-01:
I will take a look at this - many thanks for your report!

df8oe, 2016-09-01:
Aou were right.... One wrong decision ;)

Fixed & closed

closed 2016-09-01 by df8oe

steinhj, 2016-09-01:
Thanks
I have installed fw 1.3.0 (build Sep 1 2016 - 10:29:34) and have the following information:
1. ) Iambic_A is still broken in the same way.
2. ) Automatic Mode Selection (CW-U/CW-L) depending on band is working fine now.

Regards Stein LA2FDA

steinhj, 2016-09-01:
In addition to my last msg about iambic_A I have some info about Aut USB/LSB.
Correction. There is still something strange with CW-modes in band selection when selecting 60m. I am usingCW-freq offset: Aut USB/LSB
When I cycle through the Band+/- I get: 
160m cw-L 
  80m cw-L 
  60m cw-U    !!! Not able to change it to CW-L except from selecting cw.freq.translate LSB / LSB DISP / LSB SHIFT
  40m cw-L     
  30m cw-U
  20m cw-U
  15m cw-U
  12m cw-U

!!!!! IN SSB-MODE (voice) ALL WORKS FINE !!!!!

df8oe, 2016-09-01:
Hello,

on 60m you have to use USB - NOT LSB!!! So CW is CW-U -that's correct. But if you want to change, you canpress & hold mode button and then it togles to the "forbidden" mode.

Iambic A works for me fine now - I cannot reproduce any fault.

For demonstration:
Set WPM to  8 or 9.

IAMBIC_B:
If you release paddles (both pressed!!) at a "dah" there follows a "dit".

If I do same in IAMBIC_A the last ah i completed and transmission finishes.

I have tested dozens of times: working now...

73 de
Andreas

steinhj, 2016-09-01:
1. I am not using 60m and didn't know that. Thanks. 
2. I have been switching to and from between different versions, and my previous info was probably  based on wrong version. I confirm iambic_A to work in fw 1.3.0 (build Sep 1 2016 - 10:29:34)
Thanks Andreas. You are doing a fantastic job for a lot of us !

steinhj, 2016-09-01:
Hello Andreas.
I am sorry to tell, - I still think there is an issue with the cw-lsb/usb selection. 
It does not look like it is related to the band, but to the memory location the band is stored in. 
In my previous example I told you about the 60m band. 
Now I have skipped the 160m and have my bands in the following order:
80m, 60m, 40m, 30m, 20m.......  (40m is in position 3)
On 40m when cw-freq-translate i set to Aut-USB/LSB, and I select cw-mode the sideband is set to CW-U.
In voice-mode it is set to lsb. 

I have tried to tune to 80 and 160m as well when in this "memory-position", and it is impossible to set cw-L whithout entering cw.frequency. translate LSB in the menu. (even with pressing and holding mode-button)

73, de LA2FDA, Stein

df8oe, 2016-09-01:
I cannot reproduce here.

On all bands where "LSB" is appreciated toggling thrpugh modes selects "CW-L" and on all bands where USB is appreciateted there appears "CW-U".

df8oe, 2016-09-01:
BTW:

I *never* tried to swap bands in memory order and I *never* will try that :)

steinhj, 2016-09-01:
Then I know that one of the storage places is acting different from the others, and is allocated to the 60m band. Thanks for the information.
**Case closed.**
Have a nice weekend.
73 de LA2FDA.

df8oe, 2016-09-02:
I nevere have had imagined that someone could tune to other bands on a correspondent storage place so I do not know if this causes problems in other functions. Storage places are designed for soring "their" bands. It may function - but it must not. This is a feature which will be implemented when we start implementing memory places.

But it is not so difficult - I replaced the decision "CW-U on 60m" to a frequency coupled one, too. You can test if it works to store other bands than the designed on the storage places and feedbacks are welcome! May be we do have luck and issues when doing that are quite harmless ;)

Have a nice weekend, too
Andreas

steinhj, 2016-09-02:
Hello again Andreas.
As far as I can see, there is nothing indicating which storage place I am in. After the "Fast Tuning" (which is a nice function) was implemented it is very easy to accidental jump over to another band in the same storage place. And if that happen, the only way I have figured out how to know which band storage place I'm in, is to look on the storage place beside and see what frequency it is there.  If the frequency in this place also is "screwed" then I am really out on confusing thin ice :-)

Suggestion:
* An indication which storage place I am in. (might be the easiest solution)
* Possibility to disable storage places. (if I don't have antenna for 160m, I don't really need/want to see it.)
* frequency controlled settings instead of storage placed controlled settings. 

Don't put much work in this. I know how to handle it now, but it might be other persons later on getting different problems due to this issue.

I will test your last suggestion when I am home this evening, and give you feedback as soon I have done it.

Again. Thanks a lot for a fantastic job you are doing.

73, de LA2FDA
Stein

db4ple, 2016-09-02:
The thing with the storage of out of band frequencies:
I stumbled across this issue a while ago. Was thinking to give the "original" band a different color to make it stick out if you're leaving the original band. Wasn't convinced that a user would be able to decode the meaning of the color change (right now: white -> we are in a official ham band, yellow if we are in a recognized radio band (like 49m, but out of ham band, and orange "Gen" if anywhere else) . 

Any thoughts about that / better idea?
Danilo

steinhj, 2016-09-03:
Hello Andreas & Danilo.
I installed the sept.2 version. It is now possible to use the third position for other bands than 60m.
I dont know if there is any other frequency dependencies linked to each band position which might be a future problem.

Danilo
I think the most important is that users must know how this designed and that each "band storage place" is designed for a special band.  If we know that, it should not be a problem to just use i.e, the 80 m band position for the 80m band only. 
The problem today is that we don't really know which "band storing place" we are operating in. 
I think the easiest solution is to put the "band storage place" somewhere in the display. (I suggest to do it over the actual band tuned band to the right of the frequency.) The text "Gen" if out of ham band works ok, and I don't see any need for any colour coding of the frequency. 

Thanks for a good job.