df8oe/UHSDR #561: REQUEST: Tapping on tune button with "rounding" function

DG9BFC, 2016-09-03:
hello programmers
tap on tune button sets last digits to .000 ... so far so good
but if the qrg was a few ten hertz below that .000 spot it does not sum up or round up
example qrg is 14069.980 ... tapping should round upwards to 14070 but it sets to 14069
tapping on tune should be with round upwards if 100kc digit is 0.5 megs or above ... and round downwards if below 0.5 (xxx.499 and lower)
greetz
sigi dg9bfc

df8oe, 2016-09-03:
Hi Sigi,

there was a discussion when I implemented this. Result was 0% for "cutting" and 20% for "rounding".

So only solution would be to add an extra config menu entry and an extra config storage byte. Maybe this will come up. But we must be careful not to let explode configs with tons of settings...

DG9BFC, 2016-09-03:
20% is better then 0 % ... grin
no need for a menu setting
just round up like you do with am and sync am tuning
means if you are a few hertz belo 000 then sum up
if you are above 000 then cut
guessing that nobody would moan about such a function (cause we are trying to be on 000 anyway ... right?!? ... it was issue 263 if i remember right (there you added the touch on tune button function)
and regardless if you are a few 10 hz above or below ... the set function would shift you to the next full kc
but now if you are 10 or 20hz below 000 then you are a full kc lower (see example you then are from 14069.980 down to 14069 ... instead 14070 would be correct)
....
not to hurry up with adding that ... just if you have some spare time (while developing the freedv stuff ... grin)

DG9BFC, 2016-09-03:
if last digits is 0.5 to 0.999 then round up
if last digits if 0.001 tp 0.499 then cut away
no menue entry needed ... no menue digit lost

df8oe, 2016-09-03:
That's your opinion.

As I implemented 7 OMs didn't like rounding and what cutting. I don't want to bother them!!!

closed 2016-11-09 by df8oe