df8oe/UHSDR #563: HW Info, RAM

dm6tt, 2016-09-09:
I swirched the processor to the recommended 2MB flash / 256 KB model.
The HW info shows:
CPU: 419h
Flash: 2048
RAM: 0

There is a problem with recognizing the 256kB.

73, Marcus, DM6TT

db4ple, 2016-09-09:
Hello Marcus,

thanks for reporting. Good news is, the recognition of RAM size works, only the display of information doesn't. We'll have to fix that. Should not be a big deal but may take some time.

73, Danilo

db4ple, 2016-09-10:
Please verify that next  daily build works for you. Should fix the issue. In this close, any other case please report.
Thank you, Danilo

dm6tt, 2016-09-10:
Hi Danilo,
just check the daily build. The RAM is now displayed correctly with 256K.

Thanks a lot

Marcus

closed 2016-09-10 by dm6tt