df8oe/UHSDR #569: ISSUE: f/w mcHF.1.2.0 NO AM TX Balance!

I6QIZ, 2016-09-12:
Hi all,
I’m not able to balance AM TX Mode. Everything is ok in RX but in TX I’m able to balance only SSB and FM. I’m using f/w 1.2.0 on 0.5 hardware with Eeprom 24lc1026 installed. No touch screen. I down graded again to ver. 1.0.0 to verify and the AM tx balacing is OK with it. I put back a new downloaded mcHF1.2.0 but no way, it does’nt work. I see on another side receiver (HDSDR software with EMU 0202 USB) a double modulated carrier around the central frequency. I do use -6KHz frequency translate. Tryed also +-12KHz with no success. Is it possible a software or typo issue? Or do I have to set something in config menu? Many thank for your excellent software job on this nice piece of hardware. Best regards.

Luca i6QIZ.

df8oe, 2016-09-13:
Is this still present in daily snapshot?

73
Andreas

I6QIZ, 2016-09-13:
Hi Andreas,
yes, It is. I've just tested the daily snapshot 1.3.0 but no way. Still twin peacks on AM tx. All the same as described above. Does it appen to me alone?? No one else proceeded with the config adjustements or no one is going to use AM in TX?  Neither do I.  It is just a matter of interest. I could attempt a survey on Yahoo NG to see if it happened to someone else. 
73 Luca

df8oe, 2016-09-13:
I am nearly sure nobody transmits in AM - me included :) I will have a test. Many thanks for your report.
73
Andreas

db4ple, 2016-10-03:
Ok, Andreas found the first broken build with 8c17d9d2f5818777e63c1ce30f3026b9fc4cd84d , around August 13 2016. Will now try to figure out what goes wrong from here on.

db4ple, 2016-10-03:
OK, AM seems to be fixed, but we (technically I) also broke FM TX. So it is not yet completely resolved.

db4ple, 2016-10-03:
Hello Luca,

we know fixed both AM and FM part. Please wait until the pull request is merged and Andreas has published the next daily build. Then verify that the issue has been fixed and close the issue.

Thank you!
Danilo

df8oe, 2016-10-03:
I can confirm that everything is working for me now.

I6QIZ, 2016-10-03:
Tested. Working fine for me too, good job!  Thanks a lot.
Ciao
Luca

closed 2016-10-03 by I6QIZ