df8oe/UHSDR #574: S meter time constant

shkmr, 2016-09-16:
What is the reason for very large (50 times of Attack) time constant for Decay?
I feel the S-meter is moving very slowly.

I am talking about m_DecayAlpha in ui/lcd/ui_spectrum.c.
Changing it from 0.0391 (500ms) to 0.3297 (50ms) makes S-meter response
reasonably quick and dBm display is not very busy yet.

Thanks
Shigenobu Kimura

DD4WH, 2016-09-17:
Hi Shigenobu,
to be honest: no reason, that it is 0.5 s for decay.
But the normal reason why decay is slower than attack: fast, high impulses must be seen faster than low signals . . .
You are the first to wish for a faster decay time, I could easily live with it: so, from my side: green light to alter it and push it to github. Go ahead!
Please be aware, that this only alters decay for S-Meter, if you have chosen S-meter based on dbm or dbm/Hz display in the menu. It does not alter decay for the old school S-meter.
73 de Frank

shkmr, 2016-09-17:
Hi Frank,
Thanks for clarification.
I will send pull request.
-- Shigenobu

df8oe, 2016-09-20:
merged.

closed 2016-09-20 by df8oe