df8oe/UHSDR #577: UTC clock on display

S52CQ, 2016-09-22:
Hi,

I was wondering if UTC clock could be added to LCD display? Probably additional RTC module with a battery should be added to HW in this case.

It would be a nice feature to have when going portable. There is not need to manually set clock because it could be set once it's connected to USB.

Best 73 de Jure, S52CQ

df8oe, 2016-09-22:
UTC clock will bei implemented in rf bands expansion pcb

73 de
Andreas

db4ple, 2016-09-22:
Hi Jure,

I'll kind of want to encourage you to find someone to work on that. 

I am pretty sure, the actual I2C stuff (this how many of the cheap RTC module can be talked to) is not a big deal. 

However, to get the USB to set the time, that needs some smart guys. There is no standard protocol for setting the time of an external device via USB, so you'll not only have to write your own HID device driver code for mcHF side. Relatively simple, but you'll need to change the existing code, since all hardware USB OUT endpoints are used for CAT and  AUDIO. So you'll have to switch that somehow. And you'll need to write a USB driver and application to set the time on the PC.

Of course writing all the software and finding a solution to the mentioned challenges is a lot of fun. So feel free to jump in.  
Or make the clock setable via the mcHF menu system, which for sure presents less interesting challenges but is much, much easier to do.

df8oe, 2016-09-22:
Additional:
I want to set clock via menu - not via USB :)

DG9BFC, 2016-10-27:
What about using M41T00CAP from STM??? it is a real time clock with integrated xtal and also with integrated back up battery ... you can talk to it via i2c bus and it can be calibrated
details of that chip can be found in "funkamateur 10/16" page 949

S52CQ, 2016-10-27:
very nice chip! But it's a bit expensive: http://eu.mouser.com/ProductDetail/STMicroelectronics/M41T00CAPPC1/?qs=%2fha2pyFadug37pkQjKvUapn0dk6hghjnqr0FRetNNvY%3d

df8oe, 2016-10-27:
UTC clock will be added when rf bands extension pcb is available (it will be placed on it).

phaethon, 2017-01-05:
btw, it is, also, possible to get clock from a GPS module. Clock is, also, necessary for possible implementation of JT modes (if the CPU will have the capacity)

db4ple, 2017-01-05:
AFAIR line 98 in this commit https://github.com/df8oe/mchf-github/commit/655bdb36614245a800662a38cb1afb1434e8377f#diff-d91627b795ff9335ec2d2b54354af01d

was the trick to get it working.

GPS: Yes, but for this some has to implement a driver for a serial to USB converter (or directly connect the GPS serial to a STM USART...)

phaethon, 2017-01-07:
USB middleware code is different for bootloader and core code. Is it planned to migrate it to the same (and which?) code?
I was trying to add USB host CDC from https://github.com/Joe-Merten/Stm32-Tools-Evaluation/tree/master/STM32Cube_FW_F4_V1.9.0/Middlewares/ST/STM32_USB_Host_Library (apparently taken from code generated by CubeMX) and it did not compile with existing USB host core files. What was their source/version (so that I could take host CDC files from the same place)? I could find a reference in the file.

db4ple, 2017-01-07:
We are not using CubeMX based code. We are using the "older" STM32 HAL. Exact version is difficult to say, since we merged a bit of the code. I thought, the boot loader uses more or less the same source as the main firmware, but here Andreas, DF8OE is the master of the bootloader code.

df8oe, 2017-01-07:
Bootloader is migrated to nearly the samt version as firmware do I hope that bootloader code will work in firmware, too (not yet verified). But using CubeMX is impossible.

df8oe, 2017-01-22:
UTC clock now is available - so closed.

closed 2017-01-22 by df8oe