df8oe/UHSDR #598: HRD soft control error.

BI3MEK, 2016-10-10:
i use the HamRadioDeluxe software to control the machine, when LSB (USB, AM) change to FM mode, return LSB (USB, AM) mode, the machine is mute, adjust the volume is invalid.

db4ple, 2016-10-10:
Hi,

please help us with some additional information:

Version  (git commit id) are you using: ?
Version (git commit id( you first experienced the problem with?
Did you build the firmware by yourself?
How to reproduce? -> Is there anything else to do in addition to your description to reproduce the error?
Does this always happen if we follow the instructions for reproducing the problem or does it happen only sometimes?

Thank you!

BI3MEK, 2016-10-10:
I use firmware1.31.
Download from this website.
 HamRadioDeluxe ver 7.2
using FT-817 protcol.
 reproduce:
1.LSB
2.FM
[image]

the machine is mute, adjust the volume is invalid.

db4ple, 2016-10-10:
Thank you for the quick answer. We will look into this. First step will be to reproduce the issue (I will not be able to do this until mid of this week).

df8oe, 2016-10-10:
I do not have HRD (because of I am running LINUX). But I have tested with flrig and fldigi.

I can toggle all modes - audio is always present. Of course if FM is choosen, squelch is active so no audio to hear. But if I change back to any other mode audio is present. This is normal behaviour...

df8oe, 2016-10-10:
Additional information: the second encoder is "sql" in FM...

BI3MEK, 2016-10-11:
Sorry, this problem is tested again today, reset the HRD settings, and no more errors.

73

closed 2016-10-11 by BI3MEK