df8oe/UHSDR #599: REQUEST: SWR graph

JRatten, 2016-10-13:
Hi, I am looking at potentially building a mcHF transceiver in the near future, however, I was wondering if there were any plans for implementing an SWR scan and graph function. This would be extremely helpful when taking this radio hiking or as a minimalist setup as it would alleviate the need for an external antenna analyser if deemed necessary for the antenna setup used. Are there any reasons against such a function? I apologise if this is the wrong place to be asking such a question.

Cheers,
Josh
VK3HJR

df8oe, 2016-10-13:
This was discussed before.

It is *not* planned to implement such a feature.

Reason:
There is no detection for SWR using very low TX power in hardware of mcHF. You have to tune with >= 5W - that is not HAM spirit.

So we do not implement this.

There are interesting toos like MIN60 or SARK100 which can do this using only a few mW of power. We recommend to use such a device instead of adding QRM to bands.

73 de
Andreas, DF8OE

JRatten, 2016-10-13:
Ok, understood.

Thanks

closed 2016-10-13 by JRatten