df8oe/UHSDR #601: in fm we have squelch ... pse add similar to freedv mode

DG9BFC, 2016-10-16:
if you use the freedv software there is a slider (fuction like a squelch setting) ... could we add such a function to the freedv decoding?? ... users reported that freedv signals are only decoded when signals are strong (over s9 or so) ... with a user setable squelch (say audio is switched on over a setable "ber" level) maybe we can get not so strong signals to be decoded???

db4ple, 2016-10-16:
Hi Sigi,

yes, you may be right. We should try that. I think for that we should reuse the already existing FM Squelch box (but of course with a different value).  Not if this would fix the issue S9+ issue, though. I am more inclined to think we should do a little more audio processing than we do right now.

DG9BFC, 2016-10-16:
try both (not at once) ... a bit more processing helps with lower signals to be decoded ... and a squelch setting can clean out if there is some garbeled signal (not clean enough to decode)
it should be on the user end to decide when it is silence or if he wants to hear even short fractions of a word ... that means a user setable level when audio is silent and when not
the bit more processing would sure help to lift up not so strong station over the decode border ... but a user setable limiter also helps a bit (should be setable down to lowest setting and then complete garbage would come out) ...
lets try both things in a god mix ... grin

df8oe, 2016-10-16:
Audio in freedv is not muted when you do not hear anything! If maths can nothing identify what can be decoded there is no output of decoding stage. Sometimes there is output because of some fragments are mistakenly identified as "decodable fdv signal" - so you hear garbage. But this is by chance and a result of randomly fdv identification :) So there is no "level" which can be adjusted (to get more audio at weak signals).

DG9BFC, 2016-10-19:
hmmm ... but what does the "squlch" slider do in the windoze software??
there must be a descision if that is audio ... or garbage (ber level maybe???)
or random noise could produce audio

df8oe, 2016-10-19:
Maybe it is simple test for continuous decodable signal. If percentage of non-decodable audio exceed "squelch-limit" audio is muted.

df8oe, 2016-11-09:
I mostly do have "silence" when listening in fdv. Same as listening in FM with squelch.

Sometimes I can hear something for a very short time - in fdv but also in FM when some signal passes squelch by chance.

Fdv do have squelch already - what should be added?

DG9BFC, 2016-11-09:
yes i know that fdv already has a squelch function (mute when not decodable)
... i thougtht about a user setable "border" (ber level or percentage of decodable voice chunks) like it is done in the pc gui
when set to lowest level the fdv should try to sync on EVERY signal while when set to higher border it should only lock on clean signals above a given level

closed 2018-01-30 by df8oe