df8oe/UHSDR #602: AFC is possible

Jabieraru, 2016-10-18:
Dear sir:
Is possible implement the automatic frecuency control on the mcHF? If very very util for work the mcHF with transponders for work satellite....
72,s de Jabi, ea2aru.

DG9BFC, 2016-11-06:
you should use a tracking software and set the doppler offset via cat ... that works "rock steady"
(tried with a different sdr and using dde from orbitron as tracking soft)

Jabieraru, 2016-11-06:
Ok. Non problem. Many tnx de Jabi, ea2aru.

Enviado el: domingo, 6 de noviembre de 2016 18:58
Para: df8oe/mchf-github <mchf-github@noreply.github.com>
CC: Jabieraru <petorrin@euskalnet.net>; Author <author@noreply.github.com>

 

El software de antivirus Avast ha analizado este correo electrónico en busca de virus.
https://www.avast.com/antivirus

closed 2016-11-09 by df8oe