df8oe/UHSDR #609: FreeDV

G4IOG, 2016-11-01:
I am using code 1.3.1 (last Saturday's daily update) and boot 2.0.0. When listening in Freedv mode (I have not heard anyone yet) I see the green LED on the front of the transceiver is flickering ( or flashing on/off several times a second). Is this normal or is it a bug? All other modes the green LED is on solid.
Thanks for great code the radio is working really well.

steinhj, 2016-11-01:
This is normal.
I recommend this site to find freedv qso's ( Not always anybody there ) 
http://qso.k7ve.org/

df8oe, 2016-11-01:
Hi,

that is a "feature". Because of FreeDV is beta stage we needed some informations about CPU load on runtime. So we used green LCD for indicating CPU load. The more it is lit the more CPU is busy. We will switch that off when we are sure everything is running fine.

closed 2016-11-01 by df8oe

G4IOG, 2016-11-01:
My thanks to you for the very fast replies. I understand.