df8oe/UHSDR #612: transmit cd27 band, 11 meters 40 channels

AngelSerra, 2016-11-09:
transmit cd27 band, 11 meters 40 channels

DG9BFC, 2016-11-09:
sending outside hambands already is included ... or do you want the cb channels stored???

as you surely know it is forbidden to use the mchf on 11m band (except when using it as a test transmitter on a dummy with lowest power setting)

the transmit on 11 works ... but you should not use it on the air!!!!

the extended tx range was added to use the mchf with a transverter but NOT to use the mchf on the chicken band

df8oe, 2016-11-09:
It is not legal to transmit on 11m using hardware which is not approved by law. So we do not implement further tx functions than it is already implemented.

Ham operators are not allowed to transmit "on air" using mcHF! And all others are not allowed, too :)

AngelSerra, 2016-11-10:
But in Spain a new telecommunications law has entered and therefore in Spain the licenses established in the National Utilization Note UN-3, of the National Frequency Allocation Table, are not processed, if it is legal to transmit 4 watts AM/FM in The 40 legal channels, and 12 watts in SSB, does not mention any harware, state official bulletin in boe BOE-A-2014-4950.
EB5HDT

DD4WH, 2016-11-10:
Hi Angel,

I think we need not discuss the different legislations in different countries (that will also change from time to time). Everybody is responsible himself to use the mcHF exclusively for legal actions.

You can use the mcHF for CB bands, if you want to and it is your own responsibility to respect your national legislation, if you do so.

But I will not support a software development that will give the user the impression that you can use the mcHF for TX on the CB bands legally, because in the majority of the countries where the mcHF is being used, this does not conform with the legislation.

Looking at it from a technical point of view, I do not see the need for any modifications, because the "channels" are already there, if you adjust to the right frequency step and AM, FM, SSB also works and you can measure the output power to the right level. So I do not see any point to alter the firmware in any way.

But again, in most (if not all) of the countries, TX on CB bands with the mcHF is not legal, so do not use the mcHF in that way. That is totally in your responsibility and nobody can be made responsible for that apart from yourself.

Frank

closed 2016-11-25 by df8oe