df8oe/UHSDR #613: Wrong sideband in freedv??

DG9BFC, 2016-11-09:
when i have selected digital L ... and switch off fdv ... then it seems as the mchf does receive in usb and not in lsb (just checked with a test transmitter)
bug??

DG9BFC, 2016-11-15:
if I have selected digital L and switch OFF freedv ... the rig should switch to LSB
if I have selected digital U and switch OFF freedv ... the rig should switch to USB
so only then it is possible to switch from analogue voice to digital voice and back (yes i know i can select usb or lsb with the mode button ... but that is not the point ... point is when switching digital off it should switch to correct sideband WITHOUT THE NEED TO USE MODE BUTTON several times to toggle trough the modes till you have reached lsb again
digital l and digital u should also select correct part of the spectrum .... then u and l should also shift the "green line" (receiving window) to left or right from carrier point (that green line has to be below the freedv signal to decode ... but when somebody transmits in lsb then that window is below the carrier point and in usb it is above ... AND THAT IS ALSO  IN DIGITAL VOICE ...
so ... should not a switching from digital L to digital u (and back) also shift the receiving window and so that "green line" below spectrum???

db4ple, 2016-11-19:
Hi Sigi,

I am not sure I get it correctly, are you using the touch screen to switch or the mode buttons?
Easiest is if you just list the things you do (Press this and that) and then tell what happens versus what really happens (just use the description in the first entry and add the UI actions you did in the text. 

Thank you,
Danilo

db4ple, 2016-11-19:
Hi Sigi,

tried it. And it seems to me, that enabling FreeDV always switches USB no matter, what you see. Also the demodulated SSB is USB, independent of DI-L or DI-U.
Well, the green you are refering to is not green in my case. Took me a while to figure out what you meant. It is the bandwidth indicator. By default this is gray. Better not use color names when it comes to things which are configurable in color. The BW indicator is in line with the mode indicator. The only thing not being in line is the actual demodulation. Needs some investigation what goes wrong here. Should be to hard to find and fix.

73
Danilo

db4ple, 2016-11-19:
Hi Sigi,

the code was doing exactly what you found: In Digital modes, audio was always decoded as USB. FreeDV itself was doing it correctly, so only the audio monitor was broken. Now fixed.

db4ple, 2016-11-20:
Hi Sigi,

please verify that everything is okay now and close issue.

Thank you!
Danilo

DG9BFC, 2016-11-24:
i think it is ok now ... did not test with dv on if it is then also working cause i had to make a test setup witha second trx ... (and i was out of home for holidays for a week) ... now back atr home ... fired up new fw and when switching dv off it is ok (dig l decodes lsb and dig u decodes usb when dv is off) ... if i have some time in the next days i can try a test with dv ON and then we will see :-)

df8oe, 2016-11-26:
fixed.

closed 2016-11-26 by df8oe