df8oe/UHSDR #615: Wrong sideband in CW with cat control

steinhj, 2016-11-17:
I am running cat ctrl in sandbox. (mchf-bin 1.4.0)
1.)
* I select a cw frequency (i.e. 80m band) by using cat control
* The bw indicator indicates LSB ( to the left ) and the Display shows CW-L
* I f i tune in to a station using the scope, I can see the station in the upper sideband, and still here it. If I adjust the frequency so the scope indicates the station just over the BW-indicator, - I can not here it any more.
2.)
* If I press and hold Mode (Changing between CW-Land CW-U,) the indicated station moves left on the band scope. First time I do this, there is no change in the mode-indicator(CW-L), just  a "click" and I can see the station is moving left.
3.)
* If I change bandwith (BW) after point 1, the bandwith indicator jumps over to the right side(USB) and the mode Indicator change to CW-U.

Hope you understand my way to try to explain the problem. :-)

Thanks for doing a very good job for a lots of amateurs.
I know you are sacrificing a lot of time, and I really appreciate this !

73, de LA2FDA

df8oe, 2016-11-18:
I guess you have started CW via CAT using the wrong mode (USB instead of LSB) and you have confused display by doing so... What happens if you start CW via CAT selecting the correct mode?

73 de
Andreas

steinhj, 2016-11-18:
I think thats what I am doing. I am on 40M and all looks ok, (CW-L and match between observed signal which is in on the left side and the BW-indicator, - also just under the heard signal)
Then I using CAT to go down to 80M. Now the mode is still indicating CW-L,
The BW indicator is under the scope on the left side, but the observed and heard signal is on the right side. 
Looks like it is the same problem if I switch freq inside the same band to.
I will try to investigate what my computer is sending.

db4ple, 2016-11-18:
Hi,
no matter what your computer is sending, the display should match the currently used mode. I will investigate how this can happen. Independently you should verify what your computer is sending.

Danilo

db4ple, 2016-11-18:
Hi,

please check if #616  will fix the issue.

steinhj, 2016-11-18:
Hi. Thanks for taking care.

I have now tested last daily build. The behavior is different. Now there is consistent between between the Mode-indicator, the location of the BW indicator and the heard signal, - but
I have a station on a known frequency on 80m. (3503.0) (known from reverseBeacon net.)
     * I CATControl to it. 
     * I hear it at once
     * The frequency indicator shows correct frequency.
     * **BUT**:  The Mode indicator is showing CW-U,
     * **AND**:  The BW indicator indicates USB, and the scope indicates the signal in the USB area.

Conclusion:
The radio always switch over to indicating CW-U when using CAT-Control, regardless of which band I am in,  even if I switch frequency inside the same band.  

Stein

db4ple, 2016-11-19:
Hi Stein,

that behavior you described is the intended one. That comes from the fact that the FT 817 and its CAT protocol work slightly different from the mcHF user interface. CW always switches to one sideband, and for CW in the opposite sideband, there is CW-R. In the case of mcHF CW matches (you guessed it) CW-USB and CW-R equals CW-LSB. Your CAT program will most likely allow you to send CW-R. 
If your CAT program does not allow you to send the CW-R band command, you can't use CW-LSB on the mcHF with CAT control unless we change the whole way it works.

The bug was that we did not update the UI correctly, when the CW sideband switched from LSB to USB (or vice versa) via CAT.

73
Danilo

steinhj, 2016-11-19:
Hi Danilo.
Many thanks for the explanation, and for acting so fast solving this case.
73, de Stein, LA2FDA

closed 2016-11-19 by steinhj