df8oe/UHSDR #622: 160m band not working in new fw?!?

DG9BFC, 2016-11-24:
title tells all
fired up new fw ... and i have a non working 160m band

db4ple, 2016-11-25:
Hi,
cannot confirm. RX and TX do work for me (build from github active-devel minus very latest commits which should have no impact). Please be more specific what does not work (frequency change, listening to a signal, in mode x, ... )

TU & 73
Danilo

df8oe, 2016-11-25:
I cannot confirm, too. Everything is working as before.

DG9BFC, 2016-11-25:
ok ... listening to a stn on 80m band (a pa stn on 3630)
pressed bandm to 160m band
stn from 80m still be heard ... spectrum did not change
all digits in orange (while turning vfo they change to white but flip to orange agn when i stop turning)
even tuning in 5 or10 kc steps upwards to 80m band did NOTHING (means stn that i heard in first step is still there) ... then tuned back down (with the 5 kc steps) to zero and then it shows 999 megs
seems as in my rig the 160m band is not working at all (no communication to dds chip) ... i can tune up or down on that band and it does not change the rx frequency (it seems locked on the last 80m qrg)
hope that describes it ....
is there a reset to be done?? did you make a major change in fw?? (during last week while i was out of home for holiday)

DG9BFC, 2016-11-25:
ANDREAS gave me the right tip
i resetted all settings and now 160 works again :-)
when tuned to 160m (somewhere between 1810 and 2000) it showed GEN in display (band decoding part of the code did not work) and it did not show 160m ... so it seems as my band stacking memory was a bit of a mess (i used the mchf several times out of band for listening to radio stns and maybe somewhere i made a change that i can not remember) ...
ok ... back to other things ... issue closed

closed 2016-11-25 by DG9BFC