df8oe/UHSDR #625: switching off dv does not show ssb demod

DG9BFC, 2016-11-26:
title tells all
if i switch from analogue ssb to digital voice the rig shows dig -u (or dig l) ...
but when switching off it does not show usb or lsb
pse correct this so you can then switch dig-u/usb/dig u (with tapping on touch screen)
or you can the switch lsb/dig-l/lsb/dig-l ... (also with touching on screen)
then it would be easy to compare analog versus digital voice within a qso
tnx sigi dg9bfc

db4ple, 2016-11-26:
Hi Sigi,

this is intentional. USB and LSB reception in digital modes is only for monitoring purposes. It is not intended as full USB/LSB mode. Use short press on Mode button to switch straight from Digital to Analog.

To be clear: this is how it is implemented. Of course we can change that but I personally don't think this will improve things. Other opinions welcome!

73
Danilo

DG9BFC, 2016-11-26:
it will improve usage (copare analogue and dv with a touch) ... easier in my view
yes i can use the mode button ... does the mode button switch direct to correct sideband/mode?
if i remember right you have to scroll trough all modes so no direct analog digital switching and back is possible ... so why not implement that?!? ...
the non touch screen user (yes i know there are a few) can toggle trough the modes ... but for the touch user it would be easier to switch dv on and off with touch ... and also mode then should show dig l/lsb/dig-l (or dig-u/usb/dig-u) ...
i just find that it would be LOGICAL to behave that way

df8oe, 2016-11-29:
Pse look at the actual daily snapshot. Is this the way you want to have it? Hope we have understood what you are meaning...

Test / comment / close please :)

DG9BFC, 2016-11-29:
YEPP EXACTLY :-)
switching off dv and it shows usb (or lsb) ... so with one finger touch you can switch dv off and on and off as often as you like ... that way it is easy to compare dv against analogue voice (with the actual on that moment existing conditions)

closed 2016-11-29 by DG9BFC