df8oe/UHSDR #645: USB Power Mod

M0FDA, 2016-12-11:
I would like to suggest an alternative version of this mod I have used on my V0.4 board.  All mods are on USB connector side of the UI board.
1.  Isolate the 5V from the large USB connector by cutting the track to the via next to the "P1" label (I used a 1mm drill to remove the copper from the top of the via).
2.  Clean a small length of solder mask from the track and tin with solder.
3.  Locate the 8V track which runs to R4b, clear a small length of solder mask from the track and tin with solder.  Note: this track runs quite close to the USB +5V track isolated above.
4. Connect the input leg of a 78L05 voltage regulator to the +8V track.
5. Connect the output leg of the regulator through a diode (any type rather than a low voltage shottky type) to the track to the large USB connector.
6. Connect the ref leg (centre pin) of the voltage regulator through a diode of the same type to ground.
7. Connect the output leg of the regulator through a third diode of the same type to the +5V pin of the DFU USB connector.

This modification ensures the supply to both connectors is at 5V (no diode drops to be taken into account), and the two ports supplies are isolated from each other, while the regulator acts as the current limiting element.  
The bias current (1.25mA) through the ref pin will forward bias the the diode lifting the reference pin the one diode drop above ground.  This will result in the output of the regulator being one diode drop above 5V and hence the supply to the two ports being held at 5V.
I hope this suggestion is useful to you.

df8oe, 2016-12-11:
Small USB plug does not need 5V supply since a few months. Of course it is better to get own power supply or USB (and it is already implemented in 0.5). I never have had problems running 0.4 in original so never have had thought about that.

73
Andreas

M0FDA, 2016-12-15:
Hi Andreas, 
My main concern was with the fact the USB power lines were connected directly to the 5V internal supply and could be back fed by an external device being connected, bypassing the ON-OFF switch (which, technically, only controls the 8V regulator).  Also, the "official" mod required the diode to be fitted under the display, which is difficult for those who built the board and soldered in the display as I had.
 I have since socketed the display, by the way.

Vy 73
Alan

closed 2016-12-18 by M0FDA