df8oe/UHSDR #647: Audio peak

oz8gw, 2016-12-11:
Hey.
I was glad yesterday when I saw the new SW with the problem of the audio peak when the PPT was activatet solved.
Sorry to say, but I can hear no difference in a RX just beside the McHF.
I have switched software from the 8. and 10. of december several times, but no difference.
Vy 73 de OZ8GW, Leif.

df8oe, 2016-12-11:
There are many different causes for an audio peak, and the peak(s) you hear are build out of more than one cause we investigated.

I have fixed the "very big peak" which is seen on newer WM8731 when mic preamp is switched on. Many other reasons are possible active at some mcHFs. All my mcHFs do never show such a peak...

Please can you test if the peak is seen if you transmit in audio mode "line", too?

We are investigating and gathering informations at the moment.

73 de Andreas

oz8gw, 2016-12-11:
Hi again
Thanks for the answer.
I have tested the Mic/Line Select. The peak is only at the signal when MIC is selected.
I have installed the new SW - dec. 11. The peak is now gone when the TX Mute Delay value is more than 0.
Thanks for your good work.
Now I hope you will find out bug causing bad mirror-suppression at startup.
Vy 73 and merry christmas from OZ8GW, Leif.

df8oe, 2016-12-17:
Issue is gone.

closed 2016-12-17 by df8oe