df8oe/UHSDR #652: "Waterfall"

ly2bok, 2016-12-13:
When will the "waterfall" and not a "fountain" on the display?
They promised to fix. We wait.

73! LY2BOK Justas

df8oe, 2016-12-13:
I like fountain!

Reason:
If I see upcoming signal and want to tune to it using touchscreen I precisely can see frequency because signal is near to ruler. If it starts "up" falling down I must calculate frequency. Better it starts down :)

M0FDA, 2016-12-15:
I wonder if it would be possible to have two rulers.  The start ruler as currently configured (could be at the bottom for fountain or at top for waterfall) and displays the current frequency setting, the stop ruler displays the frequency and scale of the display as the display scrolls off screen.
This would allow the user to read the position of the signals at either end of the display, which can be different when scrolling through frequencies, how many of us tune by watching a strong signal line creep across the display, only to have it disappear just before we get to it as the other station we can't hear replies.
This would require the software to track two variables (the start frequency and a step, or scale size) for each line of the waterfall/fountain, and the direction of the display could also be selectable to suit users preference.

73
Alan M0FDA

df8oe, 2016-12-15:
Most users do want more resolution for waterfall and/or spectrum so it is not a good idea to grab some space from it for a second ruler... My idea/vision is a "dynamic waterfall" where the speed of refresh decreases to the top of the "fountain". Newly signals are displayed instantly and if data gets older it moves slower so that you can tune to frequency even if signal disappered... But it's not "on top" of priority list.

73
Andreas DF8OE

closed 2016-12-16 by ly2bok