df8oe/UHSDR #656: Harmonic in CW

ly2bok, 2016-12-17:
[image]
[image]
This letter from LY3BD.
> We send signal from mcHF  and  see on  shcreen  SDR-1000 .You must see  this error in program.
> LY3BD Alex , LY2BOK Justas

df8oe, 2016-12-17:
Many thanks for your report. We will take a look at this.

db4ple, 2016-12-17:
Hi,
since the problem is visible even in Tune mode which for most parts is identical for all modes.
The fundamental difference between CW and SSB is that it does not use frequency translation for generating the transmitted signal. This may cause the problem.  I don't know why frequency translation is not used from CW transmit. Maybe someone else can shed some light on this.

73
Danilo

df8oe, 2016-12-17:
No. I can reproduce it in Tune and in CW "normal operation". The signals are too strong - I confirm. Is it possible to uniform signal production for CW to translated, too? Advantage is that IQ adjustment fits to CW like on any other mode.

73
Andreas

db4ple, 2016-12-17:
Hi,
of course, "normal" CW signal generated the same way as the Tune CW signal. So there must be the same harmonics. I just meant that the SSB Tune signal is generated the same way as the CW signal with frequency translation being the difference.

As said, I have not clue why there is explicitly no frequency translation on CW transmit. Adds some complexity to the whole frequency business. Could be that we would need to use different IQ adjustment settings. 

But these harmonics have been there all the time I'd say. This is nothing new.

73
Danilo

db4ple, 2016-12-17:
Hi,
I just remembered the reason for using no frequency translation in CW TX. It is the sidetone! Using no frequency translation allows to generate the sidetone from the transmitted signal. Sidetone would be lost if we use frequency translation. 

73
Danilo

df8oe, 2016-12-17:
I agree they must be present all the time. Now they are identified first ;)

I think we must add two new adjustment settings. First you have to adjust TX IQ in SSB, and then you have to adjust it for CW. I will take a look if the difference is same on low and high frequency and if so we only need one new iq adjustment point. Please stay tuned for a few minutes...

df8oe, 2016-12-17:
Not delightful at all. The spurious signals are *not* dependent to IQ settings. They are always present at same strength. The only way to get rid of them would be to change CW TX procedure. It must be in translated mode, too if signal cleanness is same as in other modes :(

db4ple, 2016-12-17:
And then there is no easy way to generate the sidetone. In fact there is none. The only way to generate audio output is the Codec DAC and it is producing IQ signals during TX.

db4ple, 2016-12-17:
Maybe it is the DC Offset causing the problems:
https://github.com/Nuand/bladeRF/wiki/DC-offset-and-IQ-Imbalance-Correction#Correcting_the_TX_module

df8oe, 2016-12-17:
So we must keep this in focus and investigate more. No solution possible at the moment...

DC correction may be a solution (and maybe can cleanup signal in other modes more, too???)

But you have to change hardware for this... Add a potentiometer to adjust by hand...

db4ple, 2017-01-03:
Hi,
please test the new CW IQ settings for TX. Should reduce the problem. See Wiki: https://github.com/df8oe/mchf-github/wiki/Adjustment-and-Configuration-Manual#iq-calibration

73
Danilo

df8oe, 2017-01-10:
closed now.

closed 2017-01-10 by df8oe