df8oe/UHSDR #659: Sidetone in CW when TX disabled

steinhj, 2016-12-18:
Earlier it was possible when disabling TX  (by long pressing tune) to hear the sidetone in CW, without any RF out. It looks like this function is not working anymore.
Daily build Dec 17 2016 11:05:39

db4ple, 2016-12-18:
Yes, this is gone, you're right. It is an unintended side effect of the recent changes. Can and will be fixed.

73
Danilo

ly2bok, 2016-12-19:
In ver.2016 12 19  no working  PTT and CW   and TEST .
Feeling that the programmer does not have to SDR MCHF and unable to check changes.
Maybe I'm wrong.  Congratulations to all team with the upcoming holidays
Justas LY2BOK

df8oe, 2016-12-19:
You are right: programer does not have access to mcHF in the moment and is doing everything "theoretically" :)

Will be fixed shortly.

Happy Xmas and good holidays, too.

ly2bok, 2016-12-19:
It's hard to work when there is no check in kind. Success to all, and good ideas
73!
Justas LY2BOK

df8oe, 2016-12-19:
Hopefully fixed all :)

steinhj, 2016-12-25:
Happy Xmas, and thanks again for the great job you are doing for a lot of amateurs.

The CW sidetone is back. Thanks, - but:
Between daily build from Dec 02 and Dec 17 something happend to the cw TX.
It is now  impossible to use a straight key, or a external keyer. It seems like the first dot or dash in each char is chopped. When using the onboard keyer it is better, but still some dots missing now and then.
I discovered this when I due to the problems with internal keyer was trying to use an external, - and it was worse. I am now back to DB Dec 02 23:01:33, which works fine.

LA2FDA, Stein

db4ple, 2016-12-25:
Hi Stein,
what is the setting of your "TX Mute Delay". The problem could be that the real CW transmission now starts much faster, but TX Mute Delay may mute this out. Please set this to 0 (in the new firmware of course).
I verified with the DSO that the straight key is working perfectly in terms of seeing the ramp up of the tone in order to prevent clicks. And this happens within a few milliseconds after switching on the TX. If you use an external PA, this may be to slow now. Challenge for me is, that I don't know CW so I can't spot problems here. However, others have tried to use CW with the new firmware  I guess and did not report the same issue. So we may have to look for some potential problem sources on your side (settings, external PA).

But I guess we can fix that this year :-)

73
Danilo

db4ple, 2016-12-25:
Hi Stein,
disregard what I have been saying related to "TX Mute Delay". It is not used in CW mute. Damn, should have known :-). So what remains is that TX now starts very quickly and the initial portion will be lost.

I think, enabling the use of TX Initial Mute Delay also in CW will solve the issue, since this setting has been made exactly for that purpose. 
In any case, I would ask to you give a more complete report of how to reproduce the problem. Please include a description of your hardware setup.
To help identifying the problem a few questions follow:
1. If you listen to the generated sidetone of the mcHF, can you hear the  "missing" dots/dashes (i.e. it is generated but not transmitted -> could be HW setup issue) ?  
2. Is the always the "first" dot or dash when coming out of RX into TX or is there also loss during the transmission itself (i.e. when it is continously in TX).
3. What is your CW TX->RX delay setting?

Maybe we should take this to the Yahoo group and ask if others had similar issues?
 
73
Danilo

df8oe, 2016-12-25:
Hi,
I have tested CW using built-in keyer and straight key. Both works fine using newest daily build. First dot is always readable. If you are running *very high* speed of course first dot is significantly shorter than following dots. But that is not an issue. Transmitter needs a few ms to become ready. All my transceivers are identical in this. If I set CW TX->RX Delay too short I get garbage - but I never want to work full bk. So I have set this to 30ms. You can hear sometimes between words - never inside a sign or a word and I think thats ok.

Because of I work using Ultimatic mode (and none of my other keyers do have this mode - only mcHF) I never use external keyer.

73
Andreas

steinhj, 2016-12-25:
I will make an test later on and come back with some answers.

df8oe, 2016-12-25:
Hi Stein

I have made a test with an OM nearby and we have investigated, that default setting of CW TX->RX Delay is too short. If you set this < 20ms firmware tries to work full bk INSIDE A SIGN - which fails - and then eats some dots...

I never have had set this to smaller values than 20ms so I never have senn this issue. Please take a try with setting 30ms and tell us the result.

73 de
Andreas

steinhj, 2016-12-26:
Sorry.
I just saw there was a new DB again. I will do some test on that one tomorrow !
I just tested the version you made xmas evening :-)
73, stein

steinhj, 2016-12-26:
Test on daily Build 25/12-2016 17:13:38
Sorry to inform you.
I have now the TX initial muting time set to 0, and the TX->RX-Delay to 60ms
I now have big problems to send a CW cq and my own callsign. ( Speed 20 to 25 wpm )
If I increase the TX-RX-Delay to 500ms, then just the initial dot is the problem.....
Is there any major change in the way this is done compared to the Dec 2 daily build ?
73, de LA2FDA, Stein

closed 2016-12-26 by steinhj

db4ple, 2016-12-26:
With the newest software you can now use TX Initial Mute Delay also on CW. Transmission will now only start after passing TX Initial Mute Delay even on CW (which was not the case before). So setting it to 0 makes it behave as before (meaning as in the 17th Dec not 2nd Dec).

You did not answer if you are using an PA behind the mcHF?
What happens if you increase TX initial muting time?

What changed (compared to Dec 2nd):
In the CW generator code itself there was no real change, see github. However, the whole TX RX switching has been changed to be faster and more accurate when it comes to timings and to generate no spikes at the beginning of the transmission. 

This seems to have an impact on the CW side for you.

However, with the current information I/we will not be able to reproduce the problem I am afraid.

73
Danilo

df8oe, 2016-12-26:
Hi Stein,

if you do have problems in CW please open a new issue ticket. Describe as good as you can what you see for an issue, what kind of keyer you are using, related delay/muting settings and the speed you are TXing. I have tested again - no issue found. We need as much informations as we can get.

73
Andreas

db4ple, 2016-12-26:
Opened #669 for our pleasure since I indeed see an issue here.

steinhj, 2016-12-26:
Thanks for opening a new ticket (#669) on this.
The daily build Dec 26, without cw-delay  works nice for me. 

Anyway, - here is a small demonstration on dec-25-build, speed 24wpm, Tx Initial Muting=0, TX->RX Delay = 60ms, Iambic_B-keyer. No external PA
In the beginning of this video, i try to give some dots, and dashes:
- The red TX lamp is activated, and I can see the meter is changing to TX-mode, but
- I can not hear the sidetone on the mchf.
- There is no power leaving the mchf. (observed on external meter)

DEMO-Video: https://drive.google.com/file/d/0B8F27iz0dgsEYjBDQnR6ckdSTFE/view?usp=sharing