df8oe/UHSDR #669: CW Start Up Delay Increased in newer firmware

db4ple, 2016-12-26:
Based on discussions in issue #659  Stein, LA2DFA reported problems when using newer firmwares (like 1.5.5) which includes changed RX TX handling to fight against spikes when switching to TX. Seems this causes trouble for CW.

See the image below. This is a recording of the generated Codec Output via USB.  The upper two recordings are new and old firmware with the same settings in Iambic B mode, single dit (TX Mute and CW TX-RX  delays set to 0, 25 WpM). As you can see, it seems that we are now having more muted sound before we start to transmit. Much longer indeed. It is now somewhat in the range of 80ms and before it was something like a few milliseconds.

You can also see the improvements:
1. switching back to RX is much, much faster now. You can identify the final "spike". This is when we switch the audio channel. This is the reference.
2. The lower 3 recordings show the TX Initial Mute Delay set to 0,100,200ms and straight key mode in the current firmware. The last sound is short because I was able to get my finger off the PTT quickly enough. 

Anyway. It seems that we have something to do here. 

[image]

db4ple, 2016-12-26:
Hi,
turns out Codec switching introduces long (busy waiting) delays.
Will have to work on that later, for now we simply skip this delay in CW.

Reduces delay to about 23ms total (from 60 - 80ms). Compared to the FW 1.5.3
(before changes in RX TX Switching / Spikes/Glitch removal)
we can see there is about 13 ms loss of RX audio because of earlier muting.
The three recordings shown below are the current firmware with my last change (80ms), the old firmware  (10ms), and the changed firmware (23ms). 

So we are getting closer. Maybe it is okay for you, Stein?  Please test.

[image]

steinhj, 2016-12-26:
Hi.
This version works VERY nice for CW !
I did some new testings on the previous version (dec25)and put a demo video on #659.
( hope this will clear up a bit..)
Again. Thanks for the wonderful job you are doing.
73 - de LA2FDA, Stein

db4ple, 2016-12-26:
Hi Stein,
thanks for the video. Although we were able to identify the most likely cause on our own, this video would have helped us a lot in case we would not have been able to see it before. Sometimes it is very hard to desribe such a problem with words so that the other side can understand the problem well.

And I learned again about the internals of the software. 

Will look into the issue in a few days again, since while we are at it, we asked the german users to report regarding CW issues, maybe there are other issues as well. 

Happy Holidays and 2017
Danilo