df8oe/UHSDR #672: Frequency Translate causing noise on TX, noticable with Tune and JT65

dalerobins, 2016-12-27:
I took my mcHF out earlier on top of the mountain, I thought I'd try some JT65 from my laptop.

No one was coming back to me, so my friend put his KX3 on and my audio sounded AWFUL from JT65 it sounded noisy and oscillating. 

I tried other bands and the same thing was happening, then I pressed the "Tune" button on the front of the mcHF and it didn't sound great either.

I just got home and set my FTdx3000 to Rx and Tx on my mcHF, if I press the "Tune" button on 40m the sound is not very clean, it has noise and oscillation.

Other bands were not as bad. 

So I tried JT65 and did a Tx and also used the JT65 Tune button, on all bands my output audio was not clean, it had noise and oscillation.

I went in to the mcHF settings and changed the frequency translate settings, each option changed how my tests sounded.

When I turned Frequency Translate OFF, the output is perfect?

I did try adjusting the Phase/Balance IQ for 10m/80m but this did not change this issue.

The firmware I am using is Ver 1.5.3.

If I have Frequency translate on the mcHF JT65 is unusable for me.

Any suggestions would be appreciated :)

dalerobins, 2016-12-27:
Just uploaded a video:

https://www.youtube.com/watch?v=FykSi4_uAYM

db4ple, 2016-12-27:
Hi,

if JT65 and especially TUNE sound awful, you've got a hardware problem, since mcHF Tune is known to be fairly good sounding. And I was using JT65 just today with no problems.

I think if frequency translate doesn't work AND tune sounds ok without frequency translate you have a problem in the mixer or clocks. Since  tune is essentially a sine/cosine pair (IQ signal), with a not working IQ mixer you may mix the base frequency with one of the two signals which results in a good sounding modulated signal with the desired frequency. But in translated mode this magic does not work. For that we need a working TX QSE mixer. There are a lot of components involved in that path, but definitely something is wrong here. It is not a problem of IQ gain/phase balance. But that you had to use a very high value on 10ms could be an indicator for an issue. 

73
Danilo

dalerobins, 2016-12-27:
Hi Danilo,

Thanks for your reply, regarding the 10m IQ figure, after speaking with a few people it seems to be the same for other users with a v0.5 board they have a high IQ value for 10m.

Yes phase/balance does not make any difference with JT65/Tune with Frequency translation turned on.
I'm going to test this radio with a different firmware, I know this issue wasn't here for me on 1.4.0.

db4ple, 2016-12-27:
Hi,
I recommend to use the latest and greatest daily build from today, see http://df8oe.github.io/mchf-github/
But of course validation if 1.4.0 does not show this problem is a good idea. However 1.4.0 does not have the new IQ calibration.

Please report back after testing.

73
Danilo

dalerobins, 2016-12-28:
Hi,

Just flashed to ver 1.4.0

If I set:

USB TX IQ Bal 5
USB TX IQ Phase 0
LSB TX IQ Bal 5
LSB TX IQ Phase 0

The TUNE is perfect on the mcHF, JT65 Tune is perfect too, the only strange thing is on JT65 if I have mcHF on above 5w and slide the volume up and down for Tx in JT65 it can make a strange sound from 80-100% Tx volume, but I think the issue with a "clean" output from the mcHF over 5w is a known issue.

db4ple, 2016-12-28:
Okay, if 1.4.0 works. Try loading the new firmware and set both 80m and 10m IQ settings to the same values (Bal 5, Phase 0) and transmit on same frequency as with 1.4.0
With this you will have on all frequencies the same IQ setting.

dalerobins, 2016-12-28:
I just flashed 1.5.3

I tried the settings I used in ver 1.4.0 but it did not work, 
I set TX IQ Balance (80m) to 25 with everything else at 0, then tried 40m USB TUNE from the mcHF and that worked fine?, I do get a slight (crackle if I listen very very carefully for maybe 1-2 seconds, then its perfectly clear)

IF I send JT65 with Frequency translation ON, it's perfect up till about 4w, but if I go above 4w it does not sound as good, but if I turn Frequency translation OFF, I can go FULL and it still sounds perfect.

dalerobins, 2016-12-28:
I've been doing some testing with my FTdx3000 and also a SDRPlay.

I have observed the following, if I have Frequency Translate OFF, I get a clean signal from the radio when using tune from 0-5w - FULL, on some bands if I have Frequency Translate ON I get a really nasty signal from the radio like some form of oscillation, if I back the power down slightly it all goes away and the signal is normal. 

What I can't figure out is this only happens on Some bands with Frequency Translate on at certain power levels. 

I have attached some pictures of 40m @5w with FT on and FT off.

It is late now and I must try sleep, but I will check tomorrow to see what bands this is happening on and what power with FT on.

Please ignore the other spurs in my images, that is just noise here and also the SDR.
[image]
[image]

df8oe, 2016-12-28:
I just have tested using newest daily build, translated mode -12KHz and TUNE / JT65. Everything works fine up to full power (10 watts on 80m). There is no significant difference between translated and non translated working.

Because of your signal qualiity is power dependent I think your PA is self-oscillating. You can check this by removing RFC8. So PA does not get power supply. Now listen to the signal on a SDR. If signal is clean now your PA is oscillating...

Second cause may be your mixer is massive overdriven. What are the vaules of your PA configuration settings FULL PWR Adjust for the non working bands?

Good luck
Andreas

dalerobins, 2016-12-28:
Hi Andreas,

Thank you kindly for testing the daily build.

I have just done a quick test for ALL the bands, the issue seems to be only on 40m and 30m

To get this to happen I need to do the following:
40m 3w+ FT ON or FULL with FT OFF
Menu 5w Power Adjust any setting over 3 with FT ON, 
Menu Full Power Adjust any setting over 7 with FT OFF
30m 3w+ FT ON or FULL with FT OFF
Menu 5w Power Adjust any setting over 3 with FT ON
Menu Full Power Adjust any setting over 5 with FT OFF

All other frequencies seem fine on 5w - FULL and they are

5w menu settings:
160 - 7
80 - 5
60 - 6
40/30 see above
20 - 14
17 - 13
15 - 16
12 - 21
10 - 26

Full Power Menu Setting:

160 - 10
80 - 10
60 - 11
40/30 see above
20 - 24
17 - 22
15 - 25
12 - 28
10 - 30

Pa Bias setting - 112, current draw with PTT down 950ma verified on Fluke multimeter and TTI Bench PSU.

df8oe, 2016-12-28:
I cannot see here any relations to firmware because I cannot confirm here with one of my mcHF's...

dalerobins, 2016-12-28:
Hi Andreas,

I think you are right, I have to look further in to this at a hardware level.
I guess FT works differently with 1.5.x so this is just becoming more obvious on whatever is wrong with my mcHF, I wonder if it's a 0.5 thing.

dalerobins, 2016-12-28:
I've taken the mcHF apart and there is some oscillation coming from the Final Mosfet.

It definitely seems to be coming from the PA section or PA drive. I have adjusted the PA Bias and checked voltages (within range)

If I go on 40m and turn the Bias down so I am outputting 3w then go to 5w power adjust as I adjust it around 4-5w there is some oscillation happening.

Sometimes if I check with one lead of multimeter on the Gate of the Mosfet and other on chassis it can "supress" the noise slightly. Sometimes it can make it worse.

df8oe, 2016-12-29:
I close this now because it is definitely no firmware issue.

closed 2016-12-29 by df8oe