df8oe/UHSDR #675: Eprom nicht erkannt

inlinerbutch, 2016-12-28:
Mit dem Daily build vom 28.12.2016 15:00:30 wird mein Eprom nicht mehr erkannt (Wrong Signatur/OB)

Mit FW 1.5.3 wurde mein Eprom noch erkannt.

db4ple, 2016-12-28:
Reverted increased speed back to original (very slow settings). Please report what HW you are using and if the boards are connected via longer than normal wires. Or if you did any modifications on I2C side.

Next build should work for you again.

Thanks,
Danilo

inlinerbutch, 2016-12-29:
I am using the MICROCHIP  24LC1026-I/SM  EEPROM, AEC-Q100, Seriell I2C, 1 Mbit, 128K x 8 bit, 400 kHz, SOIJ chip (Farnell 1971897). I don't have any modification on I2C side or long wires. Let me know please if it is better to replace it. I will test the daily and give you feedback.

Cheers,
Uwe

inlinerbutch, 2016-12-29:
I have tested the daily 29.12.2016 08:49:59 and the Eeprom is detected again. Please advise if it better to use a different part instead of the upon listed Eeprom.

Cheers,
Uwe

db4ple, 2016-12-29:
Hi Uwe,
no, I2C EEPROM is fine. We tested with higher I2C Speeds and it turns out the I2C resistors are too high for the higher speeds.  Stay tuned.

73
Danilo

df8oe, 2016-12-29:
Problem is a hardware design error. The two pullup resistors R6 and R7 for I2C-bus 2 are too high. Please fit 1K instead of 4K7 and try next daily build which has enabled high I2C-speed again. Using 4K7 makes it impossible to use higher I2C speed.

73
Andreas

inlinerbutch, 2016-12-29:
I am fine with this solution. Please add this mod (recommendation) to the list on your modification side.

Cheers,
Uwe

df8oe, 2016-12-29:
We were experimenting at the moment. If neccessary, we wil add this of course. Actual builds are working without modifications increasing EEPROM and audio codec I2C speed x 8

73
Andreas

closed 2016-12-29 by df8oe