df8oe/UHSDR #676: "Reverse Touchscreen" nicht merh im Menü

inlinerbutch, 2016-12-28:
Seit FW 1.5.3 fehlt im Display Menü der Punkt "Reverse Touchscreen" für die 3.2 Displays. 

Bei der FW 1.4.0 war "Reverse Touchscreen" noch im Menü.

db4ple, 2016-12-28:
It is still there but moved to the (from one perspective) more appropriate configuration menu, where all "one" time setup items are collected. Since you rarely change to orientation of the touch screen, this is appropriate. Display menu is meant for any setting which influence the actual presentation on screen but not HW settings like the touch screen controller.

73
Danilo

inlinerbutch, 2016-12-29:
I only realized it by the fact that the initial setup in this was reseted (maybe due the Eprom issue?!) and I tried to find this point under the display menu. I will test and confirm.

Cheers,
Uwe

inlinerbutch, 2016-12-29:
I can confirm that the  "Reverse Touchscreen" point is now in configuration menu available and works as designed.

Cheers,
Uwe

closed 2016-12-29 by inlinerbutch