df8oe/UHSDR #693: No Power out

steinhj, 2017-01-01:
Hi
and Happy New Year !
I just updated my Daily build firmware from dec 26-1630  ->TO>  1.5.7 DB Jan 1 2017 01:47:24

No power leaves the mchf. ( TX-indicator lights, but no pwr indicated by internal or external meter. )
Tested in CW and SSB.

Going back to Dec 26 fix it.

73, - de LA2FDA

df8oe, 2017-01-01:
Hi Stein,

happy new year!

I cannot confirm this - CW and SSB power out ok for me using actually daily build. We need more informations.

73 de DF8OE

db4ple, 2017-01-01:
Stein, Andreas, try changing Freq. Translation to the alternative value, 12 to 6 or vice versa.

Happy  New Year 2017
Danilo

df8oe, 2017-01-01:
TX is ok in all translation modes included translate off.

steinhj, 2017-01-01:
Hi again.
I have been out for a while. Tried to go back to old version. OK. Back to new version againg, - and the same thing: No power. Then I changed the TX translate to various values, Powered off and on again.
Now it is working !
I am sorry, but I am not able to trigger the fail again, and I don't know what caused it......

73 and HNY, LA2FDA

closed 2017-01-02 by df8oe