df8oe/UHSDR #713: sideband select does not work as it should (only if xlate set to negative value)

DG9BFC, 2017-01-05:
i tried sideband selection in sam mode
seems as it does not work like it should
it only works with freq xlate set to -12 kc (or -6kc) but not when set to +12 kc (or +6kc)

when i use positive offset (high side mixing receiving BELOW local) then the sam u and l DO NOT WORK!!

try it .. and you will see that the calculation does not work correct (search algorythm does not find zero beat)

plus ... if that function works like planned ... then would you PLEASE do not put that function in menue but on LONGER MODE BUTTON PRESS??

when in sam you now toggle am/sam/am (with long press of mode) ... better would be short press selects mode ... and long press switches to "submodes" like sam/sam-u/sam-l/sam/sam-u/sam-l ....

in ssb you toggle usb/lsb
in dv it is thesame thing (dvu/dvl)
so it also should be the same in sam (sam/samu/sam-l)

for me that would be "logical function/behaviour" of that button so hiding sam u/l in menue (config screen) is not good in my view (but i am guessing it was just to try out that function and it can be impemented under long mode button press at a later state ... right??)

greetz
sigi dg9bfc

db4ple, 2017-01-05:
AM / SAM switching has been implemented before Sideband Modes were available so it made sense to allow to switch between AM and SAM easily. Now we probably should change that. BTW, AFAIR, Frank, DD4WH, the author of the new SAM (and the old one as well) suggested just that. But before we should fix the Translate issue. For now just use -12 and you're good :-) 

73
Danilo

DD4WH, 2017-01-06:
Thanks, Sigi, for your results and suggestions! 
Issue has been solved completely with the newest daily build of today evening.
SAM now works very good in both sidebands and in lower and upper sideband mode.
SAM works in all frequency translation modes.
Long Press G1 toggles between SAM->SAM-L->SAM-U->AM (we can ommit AM, if there is the wish to do so).

DG9BFC, 2017-01-07:
ok ... so i think issue can be closed (i let it open for just a small change ... see below)
yepp ... works superb
only thing that now could be added or better say changed is that in sam_u and sam_l it still shows the bandwith of sam (both sidebands)
i would like if the marker below spectrum would show same as in usb lsb when using just one sideband (but that is just cosmetic ... grin)

DG9BFC, 2017-01-08:
closed and will reopen a new issue about bw marker in sam

closed 2017-01-08 by DG9BFC