df8oe/UHSDR #717: Multiple Languages (I18N) for UI Menus and generated Handbook (Preliminary Design Discussion)

HB9ocq, 2017-01-06:
(sorry for non-english - I'm going to translate this right away...)
>   Ich finde es schade dass der kurze, 3-Stellige Identifikator (MenuDescriptor.id) ersatzlos weg ist;
>  m.M.n. waere dieser noch zur Hilfe gekommen beim weiterdrehen auf "Mehrsprachig".
>   Ich weiss, dieser ist auf gewisse Weise redundant mit dem (leider noch anonymen) enum aller Menus.
>   Ich las dass dieser nicht mehr gepflegt war und soweit auch nirgends angezeigt wurde... naja, sei es so.
> 
:
>   Ausserdem muesste doch die "doppelte Interpretation" von const uint16_t number besser ueber 
> union modelliert werden, schliesslich ist es einmal der anonyme enum und einmal der enum MENU_KIND.
> 
>   Auch wenn wir in der (embedded-)C-Welt sind: wir koennen uns bezueglich Typsicherheit 
> schon vom Compiler helfen lassen, nicht wahr?
> 
>   -  -  -
> 
>   Zur Mehrsprachigkeit: ein einfacher Ansatz koennte nun sein, die Datei drivers/ui/menu/ui_menu_structure.c pro weitere Sprache zu vervielfaeltigen:
> 
>    drivers/ui/menu/ui_menu_structure.c  # bleibt Englisch, DARF aber auch nach ~_en.c umbenannt werden (symmetrie ist besser)
>    drivers/ui/menu/ui_menu_structure_de.c
>    drivers/ui/menu/ui_menu_structure_fr.c
>    drivers/ui/menu/ui_menu_structure_it.c
>    :
> 
> 
>   Durch eine Umgebungsvariable (in der Art von LANG) laesst sich zum einen spezifisch 
> pro make-Durchlauf das FW-Binary in der entsprechenden Sprache fuers LCD 
> bauen (vorerst NUR 1 Sprache auf's Mal im Geraet) indem die Quelldateiliste SRC nur die
> eine passende Sprachvariante von ui_menu_structure*.c "mitnimmt" (z.B. durch umkopieren
> der Datei MIT Sprachkuerzel auf den Dateinamen OHNE Sprachkuerzel).
> 
>   Zum anderen kann ich meine Scripts in support/ui/menu/*py so anpassen dass jederzeit 
> parallel ALLE verfuegbaren Sprachen "tabellisiert" und "graphifiziert" werden, naemlich auch 
> in Dateien mit der Sprache im Namen *_mdtable_${LANG}.md, *_graph_${LANG}.(gv|svg|png).
> 
>   Der Vorteil ist meiner Erfahrung nach, dass man mit diff-Ansicht im Editor jederzeit 2 beliebige
> Sprachen von ui_menu_structure*.c nebeneinander halten kann und so das Uebersetzen der 
> Texte einfach von der Hand geht.
> 
>   Natuerlich ist dies mit den heutigen 9 verteilten Menudaten-arrays redundant.
>   Kommen spaeter mehrere/alle Sprachen in die FW (e.g. Geraete mit 2MB FLASH) muss 
> da natuerlich z.B. auf 9 Menudaten-arrays * Anzahl Sprachen aufgebohrt werden (oder 
> eine cleverere Anordnung wo bei fehlender Uebersetzung "per Fallback" auf EN 
> zurueckgegriffen wird)
> 
>   Sicherheitshalber wuerde ich da das Macro UiMenuDesc(..) pro Sprache vervielfaeltigen 
> auf UiMenuDesc${LANG}(..)
> 
> 
>   Weitere, loose Gedanken dazu:
> 
>   * mehrere Sprachen in die einzelne Datei ui_menu_structure.c (z.B. per mehr Macros) zu
>     stopfen, mach das Editieren SEHR UMSTAENDLICH. Schon jetzt die langen Zeilen 
>     sind grenzwertig...
> 
>   * mit dem nun verschwundenen const char[4] id haetten wir eine Mnemonische indirektion 
>    gehabt (anstatt nur der Numerischen, ueber den heutigen anonymen enum der Menus), um 
>    die Dokutexte in weitere Pro-Sprache-Dateien auszulagern. 
>    Das waere natuerlich gegenlaeufig zur jetztigen angenehmen Praxis nur in der 
>    Datei ui_menu_structure.c zu editieren.
> 
>   * den Weg total umkehren: die Menuangaben werden in z.B. Python (oder JSON, oder...) 
>    primaer editiert; der Buildablauf generiert dann den dazugehoerigen C-Code (was heute 
>    ui_menu_structure.c ist). Je nach fuer den Builddurchlauf eingestellten Eigenschaften 
>    der Target-HW (FLASH-Groesse) nur 1 Sprache oder halt mehrere.
> 
>   * die extrahierten Texte, welche heute in der MarkDown Tabelle landen, "wiegen" etwa(*) wie folgt:
>      * gesamter Text, inkl. UiMenuDesc (= heute noch unvollstaendig) ca. 21kB
>      * nur "LABEL" Text = vollstaendig, ohne ID, ohne UiMenuDesc <4kB
>   der Fantasie, resp. Speicheroekonomie zur Frage "wo weitere Sprachen unterbringen?" ist 
> nun freier Lauf gegeben: ins FLASH damit? auch bei 0.5MB uCs? ins Config-Flash, mit was 
> fuer Ladezeiten beim Booten? wieviele Sprachen nebst EN?
> 
> (* gezaehlt hat der Computer, Bytegenau. Ich habe aber bezueglich des MarkDowns und des 
> vielen Whitespace nur genaeherte Reduktionen vorgenommen - und bisher natuerlich nur 
> ASCII, kein UTF-8 oder Unicode... auch mit packed strings liesse sich noch etwas Platz rausholen...)
> 
> Gruss, Stephan HB9ocq

db4ple, 2017-01-06:
Hi,
while I know how important internationalization is, and how attractive use of multiple language in the user interface and documentation seems for the user, there are many risk associated:

Each change in documentation then needs to be translated into multiple languages the change contributor typically will not be able to write/read.  So we risk inconsistency. New documentation/menu entries will only be available in a subset of languages. Which to use e.g. for the german user, if there is only english and french etc.  Keeping all the information consistent is a lot of effort.
 
In summary, I don't think the effort is spent in the right place. The technical approach, btw, is not a problem I see, just to be clear about that. The solution sketched above makes sense, but it must be embedded in the "standard" Eclipse (and CoIDE) workflows.

So my question is: Why not stick to English? What is the benefit of multilingual menu help texts? And will the benefit outweighed the problem associated? You have to see it in the long run. It is hard to get rid of something started being expected to be there (even though has been done  only in moderate quality).

In a nutshell, I don't think the effort is worth it. I would rather aim to have a good and easy to read maintained documentation in English plus strict use of only English in the UI.  And would think about extracting it in a form that it is easily machine translatable. By this we can offload the language maintenance to a machine. I know automatic translations are not even near perfect but at least everyone can trigger such a translation after each change without much time to invest. 

BTW, the removal of the menu ids was opportunistic, since these have not been maintained and used for a while now. Nothing against using these again with a proper system  how to assign the "right" id to them. The flash size is not a problem here.

73
Danilo

df8oe, 2017-01-06:
The reason for thinking about multi-language support are mainly the handbook texts than the language in running firmware. In our German discussion group there are many, many OMs who do not speak / cannot read english txts easily. That's the reason why they are discussing in our German forum and not Yahoo NG... Same is international. I get many emails from all over the world regarding mcHF and most of the users first tell that they do not understand english texts well. Many problems came up in Yahoo NG because of misunderstandings.

Of course I know it is difficult to keep it up-2-date. But if there is no possibility to create translated menu texts there never will be some. And we never know if it possibly could have worked.

So I am thinking at this time only about the possibility of creating different handbooks.

73 de
Andreas, DF8OE

df8oe, 2017-01-06:
I can imagine a possibility that contributions can be made using webinterface. Users who are logged in to GitHub can open a page where they can select menu entries which appear at a new page in english. There are recent banners selectable for different languages. If a translation is already available it is automatically put to text editing window. (If not window is empty). Users now can type in their translations and save the result. During my builds a bash script tests if something has changed and inserts text automatically.

So there are two stages:
1) the existing structure (which is untouched)
2) an only "online" reachable structure where users who definitely need no programming knowledge or special tools can insert their translations.
3) a bash-script that creates local files for each language from the web server
4) a possibility to build handbooks from these new files

I can do the server/php/editing stuff. And I can offer my own server to store the contents (GitHub is not working for this because of I need php and mysql access).

EVERYBODY can work on translations.

Handbooks are updated recently and hopefully grow in different languages

73 de
Andreas, DF8OE

HB9ocq, 2017-01-06:
I agree in all of Danilos points: maintenance costs manpower and once a feature is there expectations are triggered. Then technical implementation is merely a matter of taste than a a really hard problem :-)

Even though in my issue opening post I wrote about I18N in the transceiver, it is probably best to tackle the "handbook" part first (and maybe stop there), so to avoid dealing with tedious variant builds of the binaries so to not lock out 0.5MB FLASH devices.
 
So the first and most important work is to get some critical amount of reasonably well translated texts.

@df8oe - with all those user asking for, say, german: where are their inputs for translated texts?
I for one am ready to accept any written form of such input, being it in a dedicated thread in the I40 forum, here in GH as GithubPages or so.

In this spirit I think putting up an "empty table" (in the GH-Wiki) to solicit user contributed translations could possibly lower the entry barrier.

HB9ocq, 2017-01-06:
Possibly a vision on the long run may lay in "dynamic reloadable" UI-Texts.

Let me explain first and then try to circle the necessary preconditions on how to get there. (btw: the ideas here _may_ be of use for "channel storage" and "trx programming transfer from one mcHF to another" too)

Having dynamic reloadable texts would allow to "overload" the default UI-Texts from "some source".
The easiest "some source" from the user point of view would be reading a text file off a mass storage device (USB connected pen drive or (u)SD card).
If mcHF would be able to put an empty template file onto the mass storage, every user can then edit this file and put in his favorite texts. The user can then keep them for himself or feed them back to the developers to make them available to the world - so much about the offload of the translation manpower to the public :-)
If mcHF finds a translated text out of a (partially filled) file, it displays it; otherwise it stays with the built-in default text (presumably english).
Switching to any language then is a matter of changing the mass storage resp. THAT file on it.

## Preconditions
As obviously intelligible, the needs are
* a mass storage
  * USB or the config-FLASH
* a known format
  * a filesystem (or something in its place, for the config-FLASH, e.g. [TLV](https://en.wikipedia.org/wiki/Type-length-value) or something along)
* the menu entries must have a language independent key
  * the former `const char[4] id` _mnemonic_ clearly has advantages over a purely numeric `enum` one
* there must be an easy to use "bootstrap"
  * e.g. how would the (initially empty) "template" show up on the mass storage? surely _NOT_ by an ISP/STLink/YouNameIt programmer: that's not end user friendly! --> it must be self contained on the mcHF itself

Ok, enough-o-my-brainfarts err Brainstorming for now.